prusa3d/PrusaSlicer #1520: scale factor/size not updating each other in Object Manipulator sidebar

EXOgreen, 2018-12-25:
### Version
1.42.0-alpha1

### Operating system type + version
Win 10, 

### Behavior
**_BUG_** when I have an object (benchy, x, scale 100% size 60mm) and update either the scale factor or size values the value in the size or scale factor does not update (EX. scale factor of x changed to 200%, the size should change to 120mm, but stays at 60).  The only time the fields update is when a value in the position is changed and confirmed (clicked/tabbed away from) or the object is deselected and re-selected.

_Is this a new feature request?_
no, bug

m-Th, 2018-12-26:
This generates the very annoying effect of resetting the scale values to 100% (or to any previous scale given by the Size values). See https://drive.google.com/file/d/18DtYQEoLDNn0cD8ca0FKYcdmR40f1CF-/view?usp=sharing

Rowbotronics, 2019-01-02:
I'm having this issue on Windows 10 and Debian 9 as well.

I also find that it is frustrating that you have to select a different parameter to get the display to update. If not a live update, I would at least expect to be able to update the view by pressing enter.

bubnikv, 2019-01-02:
We will make the filed to be updated by pressing enter. I am pretty sure you don't want the field to be updated continuously. Been there, done that.

enricoturri1966, 2019-01-03:
This issue should have been already fixed by: https://github.com/prusa3d/Slic3r/commit/0ccd19f1eee9c732a05c0cac9215095de389691e

bubnikv, 2019-01-04:
fixed, closing

closed 2019-01-04 by bubnikv