prusa3d/PrusaSlicer #1524: Provide a new file for translation into different languages

AndylgTom, 2018-12-26:
**Version**
Slic3rPE-1.42.0-alpha1

**Operating system type + version**
windows 10 pro

**Behavior**
1) When changing the language (to any), the program gives an error. If you take your language and drop it into the en_US folder, then there is no error.

2) Q: when will you update the file to translate into other languages? 
Please provide a new file for translation into different languages. Now half of the program is not translated (3D editor view) or some lines can not be translated at all.

bubnikv, 2019-01-02:
Issue 1) is a duplicate of #1501, #1522 
Issue 2): @YuSanka , please regenerate the po files. Thanks.

bubnikv, 2019-01-29:
The .po files were re-generated, the errors on switching languages should be solved. Closing.

closed 2019-01-29 by bubnikv