prusa3d/PrusaSlicer #1526: Commandline parameter --datadir not working

Saij, 2018-12-26:
### Version
Slic3r Prusa Edition 1.42.0-alpha1 (64 bit)

### Operating system type + version
Windows 10 Pro 1803 (64 bit)

### Behavior
In the previous version you could tell Slic3r where to find it's configuration and the presets with the --datadir command line parameter.
In the newest alpha version, this command line parameter doesn't work anymore. I have cleared my specific cloud folder so it is empty, started the new alpha version but the folder stays empty.
Also if I copy my old settings in this folder, Slic3r doesn't read them but stick to the default directory.

#### STL/Config (.ZIP) where problem occurs
not existent

simonkuehling, 2018-12-27:
Just to confirm that this issue seems to be platform independent, "--datadir" is not working on Ubuntu Linux as well!

### Version

Slic3r Prusa Edition 1.42.0-alpha1+linux64 (via AppImage)

### Operating system type + version

Ubuntu 18.04.1 LTS (64 bit)

mkutscher, 2019-01-02:
Same issue here (Win 10), also no log output what is going on - it seems to ignore the parameter...

vojtechkral, 2019-01-02:
Yes, this is a known issue, so far in the alpha release, only the `.Slic3rPE-alpha` directory is used. This will be fixed.

rongith, 2019-01-04:
@vojtechkral, The fix doesn't work for me.
It seems the specified folder path parameter is converted to a wrong one.
slic3r-console.exe says folder PathToUserFolder\datadirParameterWithoutAntiSlashesAndWithoutDriveLetter doesn't exist.
I don't know what are your spec as a --datadir option. it seems relative to the user folder. I would like to set an absolute path to a folder on my D: drive.
For information, original slic3r had --DataDir as an option in the documentation, not --datadir. --Datadir doesn't exist anymore as an option. The change is not important but I tell it to clarify.

6ill9re9, 2019-01-05:
yeah still didn't manage to make it work... but mklink works fine with win 10 for me and it's closer to what I do in MacOs btw...

supermerill, 2019-01-05:
It works for me on windows.

vojtechkral, 2019-01-07:
I can confirm there still an issue, although this is a different one, probably this is about argument parsing on Windows. We'll look into it.

vojtechkral, 2019-01-07:
I can confirm there's still an issue, although this is a different one, probably this is argument parsing in general being broken on Windows. We'll look into it.

bubnikv, 2019-01-07:
So the issue is in the implementation of the C++ command line parser, which we borrowed from the upstream slic3r. Namely, the function DynamicConfig::read_cli() interprets C escape sequences, so it will remove the backslashes. The datadir will work on Windows if the backslashes are doubled.

I have to sleep over it, but the command line parser may need to be changed in respect to when and on which platforms to perform unescaping. Namely, on Unices one will likely leave the unescaping to the shell, and on Windows one will likely not want to unescape paths with backslashes.

rongith, 2019-01-08:
OK. I understand. datadir works indeed well with escape character.
IMHO, you can let as is. datadir is for experts that want to play with settings and releases. An expert can understand that and add escape character to the path. But it should be written clearly somewhere, not dug in issue discussion.

mkutscher, 2019-01-09:
> EDIT: datadir works only for slic3r-console.exe, not slic3r.exe .
Also works with slic3r-noconsole.exe, as far as I tested.

bubnikv, 2019-01-09:
This should be now fixed with 5b1c1d592261e96f626b1172601a5bd0e4348575
The unescaping is now suppressed, and a bug in slic3r.exe ignoring the last parameter was fixed as well.
Closing.

commit 237eaa5f (2019-01-03): Fix #1526 --datadir not working
closed 2019-01-09 by bubnikv