prusa3d/PrusaSlicer #1528: Crashing when cutting and duplicating

martinowren, 2018-12-26:
### Version
_Version of Slic3r used goes here_
1.42.0-alpha1 win64

### Behavior
When I cut bency ( no matter the height ). Then duplicate it. Then I want to set advanced setting for the second instance, but crashes when I click the advanced button

bubnikv, 2019-01-08:
@FidelCapo would you please test this issue?

YuSanka, 2019-01-09:
Hi, @martinowren . 

We have checked this issue. The reason for the crash occurrence is not related to cutting of an object. Crashes occur when a user attempts to add settings to an instance. 
For now you can add settings to an object or a part and not to an instance. Crash is fixed with https://github.com/prusa3d/Slic3r/commit/2dadf4b5dc153f734d4b095d62d99224dc0502c7

But we plan to implement a possibility to separate an instance from an object as a separate object. This way you could add settings to it.
This is planned to be implemented in the nearest future.

closed 2019-01-09 by YuSanka