prusa3d/PrusaSlicer #1529: Rotate infill with object

haarp, 2018-12-26:
### Version
All the way up to 1.42.0-alpha1

This is a feature request.

Currently, infill rotation is anchored to the build plate. When the print obejct is rotated around Z, infill keeps its orientation. e.g. the object is rotated by 45°, while infill is also set to 45°. This kills the stability benefits of having 45° infill.

In this instance, the user has to manually notice that infill is not rotated (usually when printing has already started :/ ) and figure out the correct rotation angle in print settings.

Also, it is not possible to have multiple objects with different rotations on one build plate with e.g. 45° infill each.

I suggest that infill be relative to each print object's rotation. Thus, setting 45° in print settigns ensures the desired 45° infill, no matter how the object is rotated on the build plate.

Thanks!

EXOgreen, 2018-12-26:
The problem is then that imported objects that don't have the correct orientation to start with and are rotated into place still have the wrong infill. Many of the objects printed at my school don't have the correct orientation orientation when brought into Slic3r. Not saying this is a bad idea, I just think that there has to be a better way to decide infill placement.

With the new alpha making it easier to access per-part-changes, would a "relative" infill angle option checkbox be suitable? That would also allow you to make a profile that has relative as the default option. This would not replace infill angle as you may need to change the default infill position.

bubnikv, 2019-01-29:
@haarp While I understand your intention, we are trying to keep the number of switches and control limited, so it is manageable not only to us, but also that the user is able to learn the software in a reasonable time. You can set the infill direction towards the print bed per object, so you can certainly achieve what you want.

How many times do you really care about the infill orientation? How many times would you use the feature you described? Will it be too cumbersome to apply the per object infill direction parameter in the rare cases you need it?

haarp, 2019-01-29:
I would actually argue that rotating infill with the object intuitively makes more sense, so it wouldn't need a new setting. It also aids consistency, as everything else is already handled relative to the object, except for infill rotation.

As for my personal use cases, I rotate infill often, either for aesthetic reasons or to change the rigidity of internal infill. With multiple objects with mutliple rotations (which might not be mltiples of 45°) on the build plate, this quickly becomes hard to manage.

I realize that such a feature might be a low priority, but I ask to not dismiss it outright.

bubnikv, 2019-01-29:
> I would actually argue that rotating infill with the object intuitively makes more sense

I bet most, if not all slicers behave the same way as Slic3r PE, that's to the intuitiveness. I understand your thinking, but I am not convinced that the world needs it.

haarp, 2019-06-27:
I keep running into this issue. When rotating objects on the build plate, for space constraints or other reasons, I keep having to remember to manually rotate infill too, to keep the same relative rotation. And this becomes a nightmare when you have multiple objects on the plater that all need different rotations to fit.

haarp, 2022-12-17:
It keeps happening. I rotated a part to make best use of Seam position = rear, but forgot that PS doesn't rotate the infill with it. Had to cancel and restart the print midway through because I noticed too late. Siiiiigh...

tastyratz, 2023-08-09:
Something worth mentioning here in the original req. is that this isn't just infill pattern direction impacting part strength. It's solid/sparse infill direction and SHELL pattern direction. If you print a part with a monotonic top surface it's going to be visually different depending on how it goes on the plate. 

I really like the idea of the per-part relative angle reference by Exogreen so I put in a req to see if Orca would at least be willing to entertain this.

closed 2026-09-29 by prusaslicer-bot[bot]