prusa3d/PrusaSlicer #1530: Tool change custom G-Code not used

Panayiotis-git, 2018-12-26:
Version:
1.41.2+win64

Behavior:
The custom "Tool change G-code" snippet is not added to the produced g-code.

TheZeroBeast, 2018-12-28:
Hey mate, not that this is a fix, it makes more sense to me if it was under tool change but the filament start gcode is inserted after purge after each change

bubnikv, 2019-01-02:
This is specific to the "single extruder multi material" setup, right?

Panayiotis-git, 2019-02-24:
@bubnikv That is correct. It is specific only to the "Single Extruder Multi Material" mode. It still happens to the 1.42.0-alpha5 version.

sufertashu, 2019-06-03:
Issue still present in PrusaSlicer-2.0.0. I can't use any custom tool change G-code.

Interestingly, the feature seems to be fixed for Slic3rPE-1.42.0-beta2.

lukasmatena, 2019-08-08:
Could you please retest in PrusaSlicer-2.1.0-alpha1 (to be released in several days) and let us know if all is ok? The custom toolchange gcode should no longer be ignored in any scenario.

screwyluie, 2019-08-22:
I don't know if the preview is wrong or what but slicing for soluble interface on the alpha doesn't appear to work... the sliced view shows everything the same as the same tool.

yes it's setup right. Even used the default soluble profile... it just doesn't work. yeah searched the gcode for a tool change and it's not there. Slicer doesn't indicate any tool changes.

Something is up.

lukasmatena, 2019-08-23:
Could you please enclose a 3MF? Note that using the custom toolchange gcode makes you responsible for doing the toolchange, so there has to be something like T{next_extruder} in the gcode in order to actually do the toolchange.
This is to allow the user to do whatever they like (which you like) and allows placing the toolchanging command in any place withing the custom toolchange gcode. This solves the dilemma whether the Tn command should be inserted before or after the custom toolchange gcode.
I'm not saying this is the issue, just that it could be. The custom toolchange gcode handling is new and can of course be bugged.

screwyluie, 2019-08-23:
>Note that using the custom toolchange gcode makes you responsible for doing the toolchange

ah you're absolutely right, I was working off the tooltip from 2.0.0 which would indicate that the tool change happens and that this code will be injected prior to it. I now see that tooltip has changed, and indicates exactly what you said.

This is fantastic actually, this helps with the issue I'm working on. I appreciate the tip. And yes I checked the gcode output, it is working and including the custom gcode in the alpha. So thanks for that.

excited now... really hoping I can use this to solve a complicated problem. Thanks again.

lukasmatena, 2019-08-23:
Glad to hear that. I'll close the issue then.

It is quite possible though that some bugs will emerge with the new MM handling, it needed quite a lot of changes. Please, report everything.

closed 2019-08-23 by lukasmatena