prusa3d/PrusaSlicer #1531: Add concentric as a solid infill candidate

haarp, 2018-12-26:
### Version
All the way up to 1.42.0-alpha1

This is a feature request.

There are some conditions under which Slic3r will add solid infill, such as near slopes, to give the following layer something to attach to. This is a good thing in most cases.

Example (touching the perimeters): https://i.imgur.com/lZEXi5w.png

However, it always uses rectilinear here, as it is the only pattern that does solid fill. As a result, printers need to do a lot of direction changes around these sections, and almost shake themselves apart. The fill pattern in print settings has no influence.

I propose that Slic3r gain another option, "Solid fill pattern", that controls how solid infills are done. Additionally, I would like to have the Concentric infill be able to do 100% fill.

Thus, such narrow solid infills like inteh example above could be replaced by concentric, which acts like additional perimeters had been added. This improves adhesion to the existing perimeters, allows for smoother print moves and less shaking, and will improve print time.

Are there arguments against this?

Thanks!

bubnikv, 2019-01-02:
I believe your proposal hinted by your image is the same as
https://github.com/prusa3d/Slic3r/issues/28

You do not really want concentric infill, but additional pieces of perimeters. It is not straightforward to implement and frankly it may even not help, it has to be implemented and evaluated first. It is something I would like to try one day.

supermerill, 2019-01-03:
> You do not really want concentric infill, but additional pieces of perimeters. It is not straightforward to implement and frankly it may even not help, it has to be implemented and evaluated first. It is something I would like to try one day.

You can test the "extra perimeter when needed" on my fork. It adds perimeters when it detect gaps below the layer. But it's very slow.

haarp, 2019-02-23:
I'm changing the title as it doesn't really represent my intention (as bubnikv correctly stated). What I'm really proposing is that in very thin sections, Slic3r should prefer doing aditional perimeters in favor of the selected infill method. The reasons being minimizing direction changes, thus improving print time, adhestion, smoothness and not shaking the printer apart.

An exampple of such a thin section:

[image]

or those added by Slic3r near slopes, as mentioned in the opening post.

supermerill, 2019-02-24:
> What I'm really proposing is that in very thin sections, Slic3r should prefer doing aditional perimeters in favor of the selected infill method.

That's relatively easy to do. Each area that are no thicker than 2 perimeter? What name to give to that, "fill thin areas" ?

> or those added by Slic3r near slopes

Well, don't use the option.

x0rtrunks, 2019-04-24:
You can achieve exactly what the OP describes in Cura using concentric solid infill pattern. It works quite well to reduce the violent shaking and reduces over extrusion in those short back and forth moves when using a larger nozzle that tends to ooze more. Also much faster due to no more rapid direction changes.

It was nice to get the option of different top and bottom solid fil patterns, but now can we get an option for seperate middle solid infill patterns? Rectilinear is usually fine, but not for every situation.

abisaz, 2019-05-09:
in the simplify3D you can choose the angles between infills
for example
Layer 1   45°
Layer 2   -45°

It would be really nice to have this option, then you could use the Line infill and put it to:
Layer 1 0°
Layer 2 0°

So it would always go paralell to the thin walls

Completly agree that this is necessary to reduce print time!

gianio, 2019-05-27:
Yes I just can agree with abisaz comment. It would really help in designing critical parts for example: sport equipment.

diabl0w, 2020-04-08:
setting infill angle to 90 degrees fixes this, but I am not sure what the other consequences to setting that angle might be

x0rtrunks, 2020-04-08:
> setting infill angle to 90 degrees fixes this, but I am not sure what the other consequences to setting that angle might be

It only fixes it in one direction for walls perfectly aligned with the infill direction.

diabl0w, 2020-04-09:
> > setting infill angle to 90 degrees fixes this, but I am not sure what the other consequences to setting that angle might be
> 
> It only fixes it in one direction for walls perfectly aligned with the infill direction.

looks, like your right... it just happened to work for the model I had loaded... so it seems the best solution is to add a setting for, "use extra perimeters instead of solid infill" and have it look for solid infill that can be replaced with maybe 5 or less extra perimeters

haarp, 2020-04-09:
> > setting infill angle to 90 degrees fixes this, but I am not sure what the other consequences to setting that angle might be
> 
> It only fixes it in one direction for walls perfectly aligned with the infill direction.

Correct. Additionally, due to #1529, this will totally fail when you rotate your object on the build plate, or have multiple objects with different rotations.

closed 2026-09-29 by prusaslicer-bot[bot]