prusa3d/PrusaSlicer #1533: Add Color change to layer edit

schneidp20, 2018-12-26:
### Version
1.41.2-rc

_Use `About->About Slic3r` for release versions_

### Operating system type + version
Windows 10

### Behavior
A feature suggession to insert color change in the layer edit, rather than have to manually edit the gcode or use prusa's online editor.   

I'm a 30+ year C++ programmer.   I haven't looked at the code yet, but I'm willing to tackle this 
if nobody is currently working on it and you accept my suggestion.   I

THanks!
David

schneidp20, 2018-12-27:
> @schneidp20 <https://github.com/schneidp20> luckily for you, the
> 1.42.0-alpha1 has this feature already! I suggest reading this
> <https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-alpha1>for
> all the changes, they have a video <https://youtu.be/qoAL6zwERg8> linked
> too, the timestamp for color changing is 6:56
> <https://youtu.be/qoAL6zwERg8?t=416>

Sweet!   Thank you!

schneidp20, 2018-12-27:
I watched the video.   Wow, you guys have done a great job.   My idea
was just to flag the layer in the old UI, but you guys brought it to a
much higher level by being able to visualize it.   Again, great job!

schneidp20, 2018-12-31:
Sorry to bug you again.   I'm sure the Q load is enormous.  Did you totally
eliminate the old way of viewing the slices?   The new 3D view is cool, but
it looses the resolution you get with the old view of the slices.   I can
kind of see what is going on if I view the slices from the bottom, but I
also like the old way of viewing the slices.   Hopefully it's still there,
but I just can't find it.

BTW: Usability wise, it would be nice to have hover-over help for adding
color marks and probably other things.  I do embedded S/W development and
recent years there has been a big focus on human center design for UI.  I'd
be willing to donate some time to the cause.

schneidp20, 2019-01-02:
> We will likely re-implement the old 2d layer view if there is enough of
> interest for it.

Re: hover-over.   I'm sure you've seen it, I'm probably just using a
different term that you use.   When you hover the mouse over a button, say
the "+" button for color change, a brief little pop-up appears with a
little text saying what that button is.   If you use Gmail, mouse over the
various items on the page and you see what I mean.

Re: Help.   Ok, I'll download and compile and let you know when I can
tackle something.

bubnikv, 2019-01-02:
I see. We use tooltips at many places.