prusa3d/PrusaSlicer #1534: Right side panel in plater slightly cropped on Linux (Ubuntu 18.04) [1.42.0-alpha1]

simonkuehling, 2018-12-27:
### Version
1.42.0-alpha1+linux64

### Operating system type + version
Ubuntu 18.04.1 LTS (64bit)

### Behavior
The controls in the right hand info panel in the plater tab are getting cropped on my Ubuntu 18.04 system. Please let me know if any additional info about my system is of interest!

Screenshot:
[image]

p.s.: AWESOME work on the new interface, guys!

BerndJM, 2018-12-27:
Looks like you have set a scaling factor for your fonts ?!?

simonkuehling, 2018-12-27:
Oh, yes, there was indeed a scaling active on my fonts - good catch!

@BerndJM, thanks for the hint, I am closing this one...

closed 2018-12-27 by simonkuehling