prusa3d/PrusaSlicer #1539: Crash when importing MMU print seperatly

danielegger1, 2018-12-28:
### Version
1.42.0-alpha1

### Operating system type + version
Win7

### Behavior
Setting:
0.20mm FAST MK3 with MMU2

Importing stls seperatly which are supposed to be imported together (as they are MMU prints) causes a crash of scli3r. The files are from prusa sample folder.

[image]

#### STL/Config (.ZIP) where problem occurs
STL=03DBenchy_-_Dualprint_-_Gunwale_Deck_Plate_Wheel_Frames_Roof_Chimney_top_-_3DBenchy.com

[MMU_Benchy.zip](https://github.com/prusa3d/Slic3r/files/2715161/MMU_Benchy.zip)

bubnikv, 2019-01-02:
@FidelCapo would you please test it? Thnaks.

FidelCapo, 2019-01-03:
Thanks for the report. I think this error is caused by enabled background processing and stl that have parts in the air. Workaround for this issue is adding all of the part of MM object at once.

FidelCapo, 2019-01-03:
@danielegger1 ok, thanks :)

bubnikv, 2021-06-03:
Do we still have this issue?

rtyr, 2021-08-03:
I was not able to reproduce it on the current stable 2.3.3 or on master. Closing.

closed 2021-08-03 by rtyr