prusa3d/PrusaSlicer #1540: Preview shape doesn't match 3D view

badger6, 2018-12-28:
### Version
1.41.2 Win64

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_Win7, 64 bit

### Behavior
The sliced preview and slices are missing chunks.
[image]
[image]
[fence gauge 2.zip](https://github.com/prusa3d/Slic3r/files/2715617/fence.gauge.2.zip)

_Is this a new feature request?_
No
#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2019-01-02:
Would you please attach a 3mf with your object? This way the object will be scaled to your resolution. 

I was not able to reproduce your issue with the current 1.42 slic3r alpha, when I just loaded and down scaled your object to fit the print bed. I suppose I would need to have your object scaled to some exact size to be able to reproduce your issue.

badger6, 2019-01-04:
I will be happy to do that.  What is a 3mf.

badger6, 2019-01-05:
closed by mistake.

bubnikv, 2019-01-05:
file menu -> save will save a project file with the 3mf extension (the 3mf is a standard for three dee manufacturing file format)

badger6, 2019-01-08:
Stll getting garbage slicing

Designspark middle cross section:
[image]

Slic3r middle cross section:
[image]

bubnikv, 2019-01-08:
You still did not attach the 3mf.
You may need to zip it to pass through github.

badger6, 2019-01-08:
[fence gauge 2 bug_0.2mm_PLA_MK3.zip](https://github.com/prusa3d/Slic3r/files/2738176/fence.gauge.2.bug_0.2mm_PLA_MK3.zip)

badger6, 2019-01-08:
OK, i zipped it and posted it.

Thank you for your patience.

closed 2026-09-29 by prusaslicer-bot[bot]