prusa3d/PrusaSlicer #1541: Erroneous travel moves when using avoid crossing perimeters

fluksh, 2018-12-28:
### Version
Slic3rPE-1.42.0-alpha1+win64-full-201812231738

### Operating system type + version
Win 10 64bit build 17134

### Behavior
Selecting "Avoid crossing perimeters" causes travel moves to outside the print area after every object

Reproduce by adding more than one item and selecting "Avoid crossing perimeters", slice

[image]

vernieg, 2018-12-29:
I noticed that too and I did not know that the "Avoid crossing perimeters" option triggers such. I simply stop using the current alpha version not knowing why the hotend moves like that.

jrdiaz, 2019-01-03:
I think that it's a duplicate of #1505

bubnikv, 2019-01-07:
Fixed with 026bc801e33778ebc5500ad0a3aac9437bca181f
Closing

closed 2019-01-07 by bubnikv