prusa3d/PrusaSlicer #1542: RepRap/Sprinter outputs unsupported GCODE and wrong toolchanging

Goury, 2018-12-28:
### Version
1.41.2

### Operating system type + version
Ubuntu

### Behavior
* Set up a printer with switching extruder
* Choose RepRap/Sprinter flavor
* Expected it to work
* M900 and M907 commands are in GCODE, those are not supported by RepRapFirmware
* It doesn't ever set temperature on tool change, resulting in printer hanging on the very first tool change.
* It doesn't change temperature even if I pick two different filament with different temperature settings.

bubnikv, 2019-01-02:
Would you please describe what you want Slic3r to do for RepRap/Sprinter firmware?

Goury, 2019-01-02:
I want this:
https://duet3d.dozuki.com/Wiki/Gcode#Section_M109_Set_Extruder_Temperature_and_Wait

So somewhere in the beginning it should put something like this:
`M109 S220 T0
M109 S220 T1`

So when T1 command is issued target temperature for tool 1 would be already set and my machine can wait for the tool to reach it (which would take zero time for this case) and continue manufacturing.

When you set temperature with M104 or M109 without T parameter it only sets temperature for active tool and other extruders stay at zero, so when T1 is issued machine waits for hot end to cool down completely and then hangs on extrusion error.

bubnikv, 2019-01-02:
Please be as explicit as possible. Please understand that we have a lot to
do, and that any bit of our mental effort that we need to invest into
deciphering your request is drawn from the time we would otherwise invest
into implementing super cool new features.

> So somewhere in the beginning it should put something like this:

At the beginning of what?

> So when T1 command is issued target temperature for tool 1 would be
already set and my machine can wait for the tool to reach it (which would
take zero time for this case) and continue manufacturing.

When? At a tool change?

> When you set temperature with M104 or M109 without T parameter it only
sets temperature for active tool and other extruders stay at zero, so when
T1 is issued machine waits for hot end to cool down completely and then
hangs on extrusion error.

When?

Goury, 2019-01-02:
K, I'll try being more explicit.

> At the beginning of what?

At the beginning of GCODE.
Right next to or even instead of temperature setting commands it should set temperature for every defined tool.

> When? At a tool change?

Yes, at the every tool change it will remain as it was set in the beginning of the GCODE.
So if you have two different filaments with different temperatures RepRapFirmware will change temperature on the only heater to the temperature of selected tool.

> When?

Whenever there is tool change command in GCODE.

And now to be super explicit, here is the example for my dual ABS setup.

What is currently happening:
1. I power on the machine
2. All the temperatures is set to zero, it's cold.
3. I make slic3 some GCODE and upload it to the machine.
4. `M190 S100` sets bed heater temperature to 100.
5. After some time bed is hot enough and machine continues processing GCODE.
6. `M104 S260` sets active tool's temperature to 260, it's tool 0 by default.
7. M116 makes the machine to wait for temperature.
8. After some tome hot end is hot enough and the machine continues processing GCODE.
9. Here goes some custom GCODE.
10. For no good reason there is `M109 S260` command, but it's okay.
11. Some more GCODE that is not relevant for this topic.
12. T0 that should be called long time ago, luckily tool 0 was already selected by default, so it'll work for now.
13. Some more irrelevant GCODE.
14.  Somewhere in section called CP TOOLCHANGE START there is T1 command.
At this point the machine will select tool 1 and will wait for its temperature (which is zero since it was never set to anything) and then fail because it is lower than is allowed for extruder to extrude.

What is expected:
1. ^
2. ^
3. ^
4. ^
5. ^
6. T0 should be somewhere before this point and commands `M109 S220 T0` and `M109 S220 T1` should be here instead of M104. Alternatively it can be `T1` followed by `M104 S260` and then `T0` followed by another `M104 S260`, so both tools would be set to 260 and tool 0 be activated.
7. ^
8. ^
9. ^
10. I don't think this command should be here, it does nothing with risk of pausing the machine for few seconds in case of slight temperature fluctuation.
11. ^
12. T0 was called long time ago. No harm could be done by calling it again for no reason.
13. ^
14. T1 will select tool 1 which was set to 260 long time ago, so the machine will make sure temperature is 260 and will continue processing GCODE.
At this point it will successfully unload filament 0, load filament 1, wipe and will continue making stuff.

How do you like this explaination?

bubnikv, 2019-01-03:
> How do you like this explaination?

It is not about me liking your explanation, but about me being able to understand your issue in the shortest possible time. Thank you, your lengthy explanation served its purpose well.

Please understand, that the day has 24 hours, and we cannot follow every possible tinkerer on Earth, who decides to implement his own firmware. The behavior you described is new to me and it differs strongly from how Marlin and similar firmwares handle the switching extruder. Your firmware seemingly simulate multiple extruders with a single extruder, while the other firmware types usually let the host do the temp switching.

BTW, do you know whether the behavior you described above is specific to Duet board & firmware, or whether there is any other board that implements the same behavior?

Goury, 2019-01-03:
It's default Duet3D board with default latest stable firmware and with recommended configuration.

lukasmatena, 2019-08-01:
I believe the issue mentioned above should be fixed by https://github.com/prusa3d/PrusaSlicer/commit/08cb5bc2c7e57424d8783ca4aa02626dd0a1a48a (the M907 etc. commands) and maybe https://github.com/prusa3d/PrusaSlicer/commit/170bd8b064d55cd580b19c597acb08d45ac35164 which fixes a serious bug in the wipe tower code that could cause skipping of temperature changes. However, if your printer firmware uses a bit different logic than is usual, it may not be enough.

Could you please retest in PrusaSlicer 2.1-alpha1 (will be released in a few days)? Thanks.

rtyr, 2020-05-27:
@Goury Is this issue still valid with current 2.2.0, can you please retest?

rtyr, 2021-01-22:
No response. Closing.

closed 2021-01-22 by rtyr