prusa3d/PrusaSlicer #1547: Ignore XY Size Compensation For MMU Prints

screwyluie, 2018-12-30:
### Version
ALL/Latest

### Operating system type + version
Any/Windows10

### Behavior
* _Describe the problem_

When slicing a multipart file for MMU, XY Size Compensation creates gaps between the colors because slic3r sees them as different parts with their own perimeters. Elephant foot compensation does the same thing but only on the first layer

* _Steps needed to reproduce the problem_

change the XY Size Compensation, slice a multicolor file and you'll see the gaps between the colors.

* _Expected Results_

perimeters of different colors should not pull away from each other or the model will not print right and will come apart. I doubt there's any way to disable it for just perimeters near each other, so it would likely be best to ignore the setting all together.
It might also be good to ignore Elephant foot compensation since it does the same thing but only on the first layer.

*_Is this a new feature request?_

Yeah I would think so.

bubnikv, 2019-01-02:
While I believe I understand your intention, I would rather keep it as it is. Adding a hidden exception rule like yours will only lead to confusion. A correct solution would be to consider a multi-color slice as one for the purpose of the XY offset, but this is a much more complex feature to implement than the current behavior.

screwyluie, 2019-01-02:
I agree that would be better but it's also not likely to happen. In the meantime prints are going to fail (I've seen a number of posted makes on thingiverse where colors are detached) and people aren't going to know why.

You get a dialog popup when you enable supports to enable the don't support bridges option, I don't see why you can't get something similar for this. I'm not saying it should be hidden, but slic3r needs to to let the user know these settings are going to break the print and cause it to fail. Something should be done.

supermerill, 2019-01-03:
> It might also be good to ignore Elephant foot compensation since it does the same thing but only on the first layer.

i don't think so. If it was ignored, the colours may be mixing / jumping on top of each other.

bubnikv, 2019-04-13:
It should be fixed in https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-beta1
Would you please verify?

screwyluie, 2019-04-13:
well x/y compensation seems to work correctly as far as I can tell, thanks!

I know we discussed elephant foot as well and perhaps that should be another post but just FYI anything more than .09 causes weird issues between the different colors on the first layer.

bubnikv, 2019-11-01:
Closing this issue as it has been fixed long time ago. We are working on improving the elephant foot further.

closed 2019-11-01 by bubnikv