prusa3d/PrusaSlicer #1548: To Build Slic3r 1.42.0-alpha1 - Prusa Edition which version of wxWidgets needs to be used?

jayraaj, 2018-12-30:


jayraaj, 2018-12-30:
Will it work if I will use wxwidget 3.0.4?

bubnikv, 2018-12-30:
We do not support 3.0.4. It may not compile, it may not work.
We have enough of troubles supporting Windows, OSX and Linuxes through one
version of wxWidgets.

jayraaj, 2018-12-31:
Actually, initially, I have used wxWidgets 3.1.2 but it was giving so many errors so i have used the 3.0.4 version then all errors are gone, But when I am debugging the code slic3r.dll is not getting loaded using LoadLibraryExW() method.

So what I can do for it?
Also please mention the supported versions of tbb library and curl.

vojtechkral, 2019-01-02:
@jayraaj See the file `deps/deps-windows.cmake`

bubnikv, 2019-01-02:
Is everything clear now?

jayraaj, 2019-01-02:
Yes, Clear thanks.

bubnikv, 2019-01-02:
Great. Closing then.

closed 2019-01-02 by bubnikv