prusa3d/PrusaSlicer #1551: MMU Preview shows different color (extruder) than 3D view

skohls, 2018-12-30:
### Version
1.42.0-alpha1+
1.41.2+

### Operating system type + version
Mac OS X 10.14.2

### Behavior
I tried to set up two multi-material objects with slightly different colors, using all five extruders.
This happens in both slicer versions and it happens regardless if I import the same files again or if I import copies of the files (so basically get a new file and a different name to avoid mixups).
I assign the extruders and the 3D view looks as expected.
The sliced preview somehow assigns the wrong extruder to my second objects wings (see files, .3mf, screenshots)

[image]
[config.ini.zip](https://github.com/prusa3d/Slic3r/files/2717168/config.ini.zip)

[image]
[MK3 MMU2_Owl_beak_PLA_0.2mm.3mf.zip](https://github.com/prusa3d/Slic3r/files/2717169/MK3.MMU2_Owl_beak_PLA_0.2mm.3mf.zip)

bubnikv, 2019-01-02:
The issue is, that the body of the owl overlaps with the wings. You may roll a dice, which of the two overlapping bodies will win.

Actually we would like to have the priority defined by the order of the parts in the object list, which would help you to fix this particular issue.

bubnikv, 2019-03-17:
Slic3r 1.42.0-beta clips the volumes in the order they are shown at the object list. 
Closing.

closed 2019-03-17 by bubnikv