prusa3d/PrusaSlicer #1552: first layer not consistent lines

vernieg, 2018-12-31:
### Version
1.4.1.2+
1.42.0-alpha1+

### Operating system type + version
macOS Mojave 10.14.2 

## Behavior
On the screenshots, you will notice different lines on the sides which also appears on the actual print. I tried this on Spiral vase and not Spiral vase option and the same look appears on the bottom layer. 

[image]
[image]

The expected output should be clean and similar to the central part of the model.

#### STL/Config (.ZIP) where problem occurs
[Archive.zip](https://github.com/prusa3d/Slic3r/files/2717679/Archive.zip)

bubnikv, 2019-01-02:
I believe this is a harmless shading effect. @enricoturri1966 what do you think?

wavexx, 2019-08-21:
The last screenshot shows a difference which is due to the winding direction of the external perimeter.
The shading is different, but incidentally the extrusion direction will also change the shading in the resulting model :)

enricoturri1966, 2019-08-22:
This issue should already be fixed in PrusaSlicer 2.1.0 alpha
[image]

wavexx, 2019-08-22:
I would close this then

closed 2020-02-07 by bubnikv