prusa3d/PrusaSlicer #1554: Arrange function produces overlapping

danielegger1, 2018-12-31:
### Version
1.42.0-alpha1

### Operating system type + version
Win7
Printer MK3 + MMU2

### Behavior
Importing twice the same model and rotate one by 180° causes a overlapping during auto center.

[image]
[image]

[MMU_Bracket_180.zip](https://github.com/prusa3d/Slic3r/files/2718236/MMU_Bracket_180.zip)

bubnikv, 2019-01-02:
> Importing twice the same model and rotate one by 180° causes a overlapping during auto center.

You mean "during auto center" or "during arrange"?

I am unable to reproduce your issue.

danielegger1, 2019-01-02:
Sorry, I forgot to mention you need to mirror the part along the y axis and yes I mean the arrange button.

[image]

bubnikv, 2019-01-02:
Sorry, I still cannot reproduce it. @FidelCapo would you please try your luck?

bubnikv, 2019-01-02:
> Just add .amf at the end of the file, maybe that helps.
>
> MMU_Bracket_180.zip

Now this fails completely.

danielegger1, 2019-01-02:
What do you mean? The app crashes?
This was exported from Slic3r with export amf file.

bubnikv, 2019-01-02:
No, it does not crash, but it does not arrange either.

danielegger1, 2019-01-02:
So there is the reported bug ;)

danielegger1, 2019-01-02:
For me the arrange works but always overlaps

danielegger1, 2019-01-03:
No Auto Arrange is disabled, when I click the Arrange button it is still overlapping parts, so that is the bug.

tamasmeszaros, 2019-01-03:
It may be that the problem lies in the amf format. Have you tried using 3mf format to save the scene and load it back? Or if you just import the model twice, rotate one and hit arrange button (**AutoCenter** being disabled) it should work fine. For me the problems started when I saved the scene to AMF and loaded it back.

tamasmeszaros, 2019-01-03:
I have found the issue, it was indeed arranging incorrectly. Thank you @danielegger1 for the bug report. The next release will include the fix.

bubnikv, 2019-01-03:
@tamasmeszaros fixed this issue with 8a3d2f385f457f9962e5a14653c2e91b73dcb6ac
Closing.

closed 2019-01-03 by bubnikv