prusa3d/PrusaSlicer #1558: unhandled xception in 1.42.0 alpha

photog0411, 2018-12-31:
### Version
1.42.0 alpha1

### Operating system type + version
win 10 64
### Behavior
This happens when I drag an .STL file onto the App Icon (to open it with slic3r PE).  It does not do this if I open Slic3r PE and then drag the .STL into the program window.

Seems Slic3r is looking for photos in a directory on my computer that it shouldn't have access to.

[image]
[image]

photog0411, 2018-12-31:
I think issue is that it's looking for them in a directory of  

*\resources\icons 

but I am just using a shortcut on my desktop and it's getting fooled.

josephduchesne, 2019-01-01:
I had mine in D:\programs and got the error complaining about being unable to load D:\resources\ from windows 10 64 bit.

I "fixed" it by running an admin command prompt and creating a symlink:
`mklink /D D:\resources "D:\programs\Slic3rPE-1.42.0-alpha1+win64-full-201812231738\resources"`

bubnikv, 2019-01-02:
It is an unfortunate bug, we shall fix it ASAP.
This is a duplicate of https://github.com/prusa3d/Slic3r/issues/1500
Closing this one.

closed 2019-01-02 by bubnikv