prusa3d/PrusaSlicer #1559: Getting Error "slic3r.dll was not loaded".

jayraaj, 2019-01-01:
slic3r.dll was not loaded error is coming when I am trying to rune latest version of slic3r.

EXOgreen, 2019-01-01:
This is not enough information to start to understand why this is happening. Please follow the template provided.

### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

jayraaj, 2019-01-01:
Version
Slic3r-version_1.42.0-alpha1

Operating System
Windows 10

I have built the source code using Visual Studio 2017.

Issue Detail:
I have built the Slic3r-version_1.42.0-alpha1 using visual studio 2017 I am able to build full source code and it's dependencies.

Now when I am debugging the source code at the time of loading Slice3r.dll using the following code.
HINSTANCE hInstance_Slic3r = LoadLibraryExW(path_to_slic3r,0,0); 
Slice3r.dll is not getting loaded. 

GetLastError() Gives error code 4520.

bubnikv, 2019-01-02:
First congrats to compiling Slic3r with Visual Studio 2017. We all use VS
2013.

If you compiled with dynamic wxWidgets, you need to have the wxwidgets dlls
inside the directory with slic3r.exe.

I recommend to use the dependency walker tool o slic3r.dll to find the
depending libraries.

http://www.dependencywalker.com/

We compile Slic3r all static, so there are no such issues. To compile
Slic3r with all static dependencies, set SLIC3R_STATIC=1 on your windows
command line before running cmake.

bubnikv, 2019-01-02:
> Thanks, Actually I have copied all the wxWidgets dlls inside the directory
> where slic3r.exe is present.
>
> Also used the dependency walker for the slic3r.dll and copied the required
> DLL inside the directory also i have compiled with SLIC3R_STATIC=1.
>
> Still, I am getting the error on Loading of slic3r.dll (GetLastError()
> Detail ERROR_NOT_APPCONTAINER(4250)).

We do not have resources to troubleshoot your issue. Googling quickly shows
that likely the cmake sets the /APPCONTAINER flag on the executable and it
is not set on slic3r.dll or vice versa.

https://www.reddit.com/r/learnprogramming/comments/5ik91a/c_loadlibrary_returning_null/
https://twitter.com/jmorrill/status/328712450067275777
https://docs.microsoft.com/en-us/cpp/build/reference/appcontainer-windows-store-app?view=vs-2017

bubnikv, 2019-01-02:
I mean we are compiling with visual studio 2013 and we do not experience
the issue you are having. We currently cannot help with Visual Studio 2017.

jayraaj, 2019-01-02:
Thanks for the quick reply.

I tried the mentioned steps but it didn't work for me.

jayraaj, 2019-01-02:
Could you please respond to #1561 because I am doing this all to get the parameters.

jayd54, 2019-01-03:
Sure.

vojtechkral, 2019-01-29:
Hi. I tested building with MSVC 2017 today and I had to modify a few spots in code (cf. a99c82340100847e56fa712f4a60c927ce4a3500) but otherwise it seems to work fine. You can follow the [doc on building on Windows](https://github.com/prusa3d/Slic3r/blob/a99c82340100847e56fa712f4a60c927ce4a3500/doc/How%20to%20build%20-%20Windows.md) to build Slic3rPE. Note that with MSVC 2017 the provided deps bundle won't work probably, you'll probably need to build it yourself (I did).

Let me know if there's a problem.

bubnikv, 2019-04-13:
No response, closing.

closed 2019-04-13 by bubnikv

jayraaj, 2019-04-13:
Thank you. For help worked for me