prusa3d/PrusaSlicer #1560: Alternate Water Soluble and PLA support layers

LawnDartLeo, 2019-01-01:
### Version
1.41.0 Win 64

### Operating system type + version
Win 10 Home

### Behavior
I swear that I had supports that were an alternate layer configuration of PLA and water soluble.  This made for easy removal of supports within internal cavities as it would just be PLA spaghetti after soaking.  I had the interface option ticked, not full.  This halved my use of WS filament.

I am now unable to reproduce.

I don't know if this was by dumb luck or if it is a current feature.

Seems to follow this https://github.com/prusa3d/Slic3r/issues/488 but that doesnt tell me anything secific on how this works... if it even is a feature.

#### STL/Config (.ZIP) where problem occurs
[image]

bubnikv, 2019-01-02:
This must have been bug, not a feature.

LawnDartLeo, 2019-01-05:
Bring back the bug!!!

bubnikv, 2019-01-08:
Sorry, too many requests, this one would likely never get implemented. Closing.

closed 2019-01-08 by bubnikv