prusa3d/PrusaSlicer #1561: Slic3r PE - multi color - how to create .gCode from console + .ini

jayd54, 2019-01-01:
Hello,
Please help me solve my problem.

I can not find any example to generate .gcode for multicolor head, using the Slic3r PE console + .ini + .stl 
I have recently bought MK3 and the multicolor head.

MY ISSUE:
I have been using the Slic3r PE from console.exe and passing the .ini file and the stl to produce the .gcode file. -- so for so good

for the multicolor, I tried to pass the multiparts of stl files, but it is not working.
Could not find any example on the web too.

Can you please help me sending any example with, a sample .ini file and the other params (including stl files), so i can generate the gcode for multicolor print.

Your help is much appreciated. 

Thank you

jayraaj, 2019-01-02:
Currently, i am passing multiple STL files like following.

slic3r-console.exe --load config.ini 3C_Lizard_1.stl 3C_Lizard_2.stl 3C_Lizard_3.stl --MERGE

This is converting the first stl to gcod,e i want that all the STL should get merged and exported to single gcode.

In GUI if we select multiple stl files we get the following message.

[image]

How we can do the same using slice3r-console.exe?

bubnikv, 2019-01-02:
If you need to do that from the command line, it is best to generate an AMF or 3MF file. Just export AMF or 3MF, rename it to zip, unpack it, analyze it, and try to generate these files through some scripting language.

jayraaj, 2019-01-02:
Can you provide command line arguments for that?Example would work.

bubnikv, 2019-01-09:
run slic3r.exe
load some files, adjust parameters
export your.3mf

rename your.3mf to your.3mf.zip
inspect the content
edit it with a scripting language of your liking
zip it back, rename it to 3mf.

run
slic3r-console.exe yourfile.3mf

bubnikv, 2019-01-09:
@jayraaj your issue is different from the original one. This one is about multi-color prints, yours is about printing a multi-color object with a single extruder?

rtyr, 2021-01-22:
No response. Original issue answered.
Closing.

closed 2021-01-22 by rtyr