prusa3d/PrusaSlicer #1562: Sliced model has errors in perimeter [1.42.0-alpha1]

Boby71, 2019-01-01:
### Version
1.42.0-alpha1
### Operating system type + version
Win10/64

### Behavior
I sliced the extruder body of the MK2s -> MK2.5 update (STL files from Prusa site) - and got some errors in the perimeter; they're just missing in some places:
[image]

#### STL/Config (.ZIP) where problem occurs
[extruder-cover.zip](https://github.com/prusa3d/Slic3r/files/2719477/extruder-cover.zip)

TheZeroBeast, 2019-01-02:
I second this, I tried to Slic3 the extruder and got the same issue on OS X 1.42.0-Alpha1

Boby71, 2019-01-06:
Udpate: This bug persists in Alpha2 as well; doesn't show up in 1.41.2

bubnikv, 2019-01-07:
This seems to be a regression caused by some larger scale refactoring.
https://shop.prusa3d.com/forum/general-discussion-announcements-and-releases-f61/slic3r-prusa-edition-1-42-0-alpha1-t26756-s30.html?sid=acc3cbcc03f545fd3cdcbe5d9704a7cf#p122560