prusa3d/PrusaSlicer #1564: MMU Slic3ing Different Temp Fix Please [1.42.0-Alpha1]

TheZeroBeast, 2019-01-02:
### Version
1.42.0-Alpha1

### Operating system type + version
OS X

### Behavior
* When you slice two different temperatures for extruders, the temperature change comes in to effect at start of ramming to new filament (this is good and logical) when going to a higher temp setting although I would logically say that the temp change needs to happen midway through purge post change so that the higher temp material can be purged before cooling. 

Attached is a Benders head that is sliced so Black and While (Ex4&5) have +10c on extruder to stop stringing with just these colours.

[eyes_middle_white_0.2mm_PLA_MK3MMU2.gcode.zip](https://github.com/prusa3d/Slic3r/files/2720236/eyes_middle_white_0.2mm_PLA_MK3MMU2.gcode.zip)

This isn't a massive issue for these small changes but will be for multi material like PLA and TPU prints.

bubnikv, 2019-01-02:
Is this behavior different from 1.41.2, or is it the same? I remember having a similar discussion.

TheZeroBeast, 2019-01-02:
I feel it is the same as 1.41.2 bubnikv, I have been testing tpu and PLA but would always get jams when PLA @205c tried to purge out tpu from 230c.

I’m not familiar with Slic3r codebase but maybe a check @ the ‘M104 S’ command before toolchange to see if next_ext_temp is lower and if so don’t insert.
Instead insert ‘M104 S’ after 3/4 or 1/2 of purge so previous filament is purged before temp drops.

I’ve also tried to get OS X compiling Slic3r to have a dig but there aren’t any specific instructions on what is needed to get this happening. Would you have any tips please?

bubnikv, 2019-01-03:
For the discussion on how to compile on OSX, please follow https://github.com/prusa3d/Slic3r/issues/1421

workinghard, 2019-01-11:
> Is this behavior different from 1.41.2, or is it the same? I remember having a similar discussion.

I think it's the same request like this here:  #1339

TheZeroBeast, 2019-01-11:
Hi working hard,

I agree that is what I consider to be the desired process for purging a hotter material. I hope this is something Slic3r can incorporate.

workinghard, 2019-01-13:
Here is a video how it works: https://www.youtube.com/watch?v=Z9pe_uwS224

TheZeroBeast, 2019-01-24:
@bubnikv any throughts on when this might be worked on?

closed 2026-09-29 by prusaslicer-bot[bot]