prusa3d/PrusaSlicer #1566: [Feature request] Make profiles comparable

Spacemarine2018, 2019-01-02:
### Version
1.42.0-alpha1-win64

One often wants to compare different profiles for filament or printer settings. Common cases could be someone sending you a custom profile and you want to figure out the changes to a standard profile. Or you change a profile that suddenly works worse than before and you want to see the differences to an older version.

At the moment there is no convenient way to compare profiles. You have to do a lot of clicking and remembering, going through each individual menu.

A solution could be a "compare mode", where you select two or more profiles and the individual settings are displayed next to each other, highlighting commonalities and differences.

icanhazpython, 2019-03-09:
Wow, crazy to me that this is not a feature yet. How does anyone remember which settings they changed vs a stock profile?

lf-, 2019-03-09:
Lock icons appear if you modify a stock profile indicating which settings are changed. Also, `diff` is a very useful tool which can be used to find those differences.

icanhazpython, 2019-03-09:
> Lock icons appear if you modify a stock profile indicating which settings are changed. Also, `diff` is a very useful tool which can be used to find those differences.

Good call, I just discovered my profiles weren't inheriting ones they are based off of. I believe it was likely an older version of slic3r that did not save this (perhaps due to a bug in that version).

rtyr, 2021-04-29:
This feature is already implemented on master. I will close this issue and leave similar  #5384 open for now.

This is how it looks currently.
[image]

Closing.

closed 2021-04-29 by rtyr