prusa3d/PrusaSlicer #1569: slic3r 1.41.2 stops responding (win 10x64)

gaelbc, 2019-01-03:
### Version
1.41.2+win64

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_Windows 10  version 1803

### Behavior
* _A "pause" symbol (looks like = rotated 90 degrees) shows up in upper left corner and app stops responding._
* _Open any STL file (I can send some if you want)_
    Place the object on the xy-plane
    Click Slice Now
    Attempt to click either Preview or Layers tab.... 
  * _If this is a command-line slicing issue, include the options used_
* _A few weeks ago it seemed to work as expected (ie the preview and layers views worked)
   Surprisingly, I can still click the "export g-code" and it will produce a gcode file.

* _It never returns_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[Rod.zip](https://github.com/prusa3d/Slic3r/files/2722560/Rod.zip)

bubnikv, 2019-01-03:
@FidelCapo would you please check this issue with 1.41/1.42? Thanks.

FidelCapo, 2019-01-03:
@bubnikv I cannot reproduce this issue with 1.41.2 and 1.42.0-alpha1

gaelbc, 2019-01-03:
Hmm, I think this one should be chalked up to DSU error.  A therapeutic reboot and nothing else running I can not reproduce the error any more either.  I suspect something else running caused the issue.

Sorry for not investigating further prior to posting.

bubnikv, 2019-01-03:
Main thing it will be fixed in 1.42 :-) Didn't you have the background processing enabled?

Anyway, nobody can reproduce it, closing.

closed 2019-01-03 by bubnikv