prusa3d/PrusaSlicer #1571: Slic3r 1.42.0-alpha1 - Prusa Edition

Glod76, 2019-01-03:
### Version
Slic3r 1.42.0-alpha1 - Prusa Edition

### Operating system type + version
Windows 10
### Behavior
Pause at random spots for filament change that I didn't add

Rowbotronics, 2019-01-03:
Maybe you added them accidentally? When you are in the preview tab and drag the slider up and down a plus sign will appear next to it. If you click that plus sign it adds a filament change pause. I know I clicked a few times before I figured out what it was doing.

bubnikv, 2019-01-04:
Would you please verify, that you did not add the color change marks yourself?

bubnikv, 2019-01-10:
@YuSanka added a nice color bar to the color change slider, which should make it crystal clear, that some color changes are there.

[image]

We believe this will handle most of the issues like yours. Closing.

closed 2019-01-10 by bubnikv