prusa3d/PrusaSlicer #1575: 1.42.0-alpha1 interface issue: Long text input field truncates when selected

bobstro, 2019-01-03:
### Version
1.42.0-alpha1

### Operating system type + version
MacOS 10.14.2
### Behavior
On long text input fields (e.g. Print Settings->Dependencies->Compatible printers condition) long text will appear to truncate when selecting the field with the mouse. The text is actually word-wrapped and can be viewed by scrolling down with the arrow keys. However, no scroll bar or other visual indicator is present 

[image]
[image]
[image]

Expect field to either scroll horizontally without truncation, or expand vertically to show full line.

FidelCapo, 2019-01-04:
@bubnikv yes, it is different. Fields work ok with 1.41.2.

bubnikv, 2019-01-29:
This seems to be a wxWidgets issue, fixed in the latest release. All we need to do is to update the wxWidgets at our build server.

bubnikv, 2019-02-01:
Fixed by patching wxWidgets 3.1.1 on the Prusa3d build server. 

Change in the input field code specific to OSX was back ported from wxWidgets 3.1.2 to 3.1.1 with https://github.com/prusa3d/wxWidgets/commit/a832410199d0930715a07feb7aff4054753d02ac

closed 2019-02-01 by bubnikv