prusa3d/PrusaSlicer #1576: 1.42.0-alpha1 interface issue: Unable to edit text using keyboard shortcut

bobstro, 2019-01-03:
### Version
1.42.0-alpha1

### Operating system type + version
MacOS 10.14.2

### Behavior
When editing large text fields (e.g. Printer Settings->Custom G-Code->Start G-code) using the keyboard select-all (CMD-A) command doesn't work. Undo (CMD-Z) doesn't work. Copy (CMD-C) and Paste (CMD-V) do work.

[image]

bobstro, 2019-01-03:
> @FidelCapo Is this behavior different from Slic3r 1.41.2?

Yes this is different. With 1.41.2 I can use CMD-A to select all text and CMD-C to copy it. CMD-Z undo does not work on 1.41.2, so there's a usability issue.

[image]

bubnikv, 2019-01-03:
It must be a new default of wxWidgets on OSX then. We shall look into it.

bobstro, 2019-01-03:
And now it's working. It must be something on my system, apologies. Working undo for text would be nice though.

bobstro, 2019-01-05:
The problem has re-appeared since loading alpha2. After a fresh install of alpha2, I click in a text field, type CMD-A and nothing happens. I then exit alpha2 and try the same thing and it works. With alpha1, the problem seemed to correct itself so I thought it was my OS, but this seems to be an issue with a fresh install.

bubnikv, 2019-01-29:
Do you still have the issue with alpha4? In Alpha4, we disabled OSX smart substitutions in the parameter input fields #830 and in OctoPrint input field #1678.

FidelCapo, 2019-02-05:
CMD-A, CMD-C and CMD-V should be fixed in version 1.42.0-alpha5. Can you confirm this? Thank you.

Patronics, 2019-02-26:
I just tested this in 1.42-alpha5, CMD-A, CMD-C and CMD-V do all work. CMD-Z still does not, however.

bubnikv, 2019-04-13:
Would you please list out the UI controls where you still have an issue with CMD-Z? Thanks.

It is likely an issue of the multi-platform wxWidgets library that we are using, so we will have to forward the issues to them, or to fix it ourselves.

closed 2019-06-08 by bobstro