prusa3d/PrusaSlicer #1580: Build plate missing in view

danielegger1, 2019-01-03:
Version 1.42.0-alpha1

With my user presets at printer profile I am loosing the build plate after reimporting the config bundle (I recently exported, started from scratch and reimported the config). There is also missing the lock icon on user presets.
Probably this bug has already been reported. I added my config bundle for debbuging purpose.

[Slic3r_config_bundle.zip](https://github.com/prusa3d/Slic3r/files/2725003/Slic3r_config_bundle.zip)

User preset
[image]

System preset
[image]

Missing lock icon for user presets
[image]

bubnikv, 2019-01-04:
The config bundle you attached was generated by Slic3r Prusa Edition
1.42.0-alpha1+win64
Would you please attach the config bundle that you generated with 1.41?

danielegger1, 2019-01-04:
I have started from scratch and created my user presets within 1.42.0
I can send you my config bundle from 1.41 but it is not completely the same (but similar)
After exporting and reimporting the config bundle inside 1.42 I loose the build plate

bubnikv, 2019-01-04:
The Prusa3D bed is currently only shown if you derived your profiles from
the Prusa3D system profiles. That means, you need to have the Prusa3D
profiles installed from wizard, and you need to derive your profiles by
editing the system profiles and saving them. This way the "inherits" field
in your derived profiles are filled with the name of the Prusa3D profile.

danielegger1, 2019-01-04:
Yes I have done that but it still looses the build plate.
So after export config bundle and reimport the build plate is gone.

bubnikv, 2019-01-04:
Would you please attach the config bundle that you generated with 1.41?
I will test your workflow.

danielegger1, 2019-01-04:
[1.41_Slic3r_config_bundle.zip](https://github.com/prusa3d/Slic3r/files/2727152/1.41_Slic3r_config_bundle.zip)

bubnikv, 2019-01-04:
I verified that there is an issue when importing a config bundle, the "inherits" flag is not processed correctly.

danielegger1, 2019-01-04:
Also the build plate?

bubnikv, 2019-01-04:
The build plate is shown for Prusa3D printers based on the references to
the respective system profiles.

bubnikv, 2019-01-07:
I dag deeper, the "inherits" flags were never imported from config bundle, so this is not a new bug.

danielegger1, 2019-01-07:
And what about the missing build plate?

bubnikv, 2019-01-07:
the build plate is shown for prusa system profiles and for profiles linked
to the prusa system profiles.

danielegger1, 2019-01-07:
Yes you have already told this but how can I link them back, that export and import config bundles is „correct“ working?
Before export they where linked build plate was shown
After import they are not linked anymore - no build plate (but I want to have it back)

bubnikv, 2019-01-07:
you cannot, we have to fix the import.

bubnikv, 2019-01-08:
Import of config bundles with the "inherits" flag was fixed with 42e8d81. Closing

closed 2019-01-08 by bubnikv