prusa3d/PrusaSlicer #1582: Command line interface in 1.42.0

jayd54, 2019-01-04:
### Version
1.42.0

### Operating system type + version
Windows 10

### Behavior

How to pass command line parameters yo SLIC3R 1.42.0, because currently whever after opening command line I am passing following parameters.

Slic3r-console.exe options file.stl

Here GUI is getting loaded it should conversion still to gcode directly but instead of it GUI is getting loaded.

Why it is happening?

How to convert stl to gcode using command line in 1.42.0

Patrik-svobodik, 2019-01-04:
Why do you want to use the command line to slice?

jayraaj, 2019-01-04:
Actually, I want straightforward conversion of stl to gcode using command line without using GUI. It was working in 1.41.2  by passing "slic3r-console.exe file.stl"

lf-, 2019-01-05:
@Patrik-svobodik I could see myself doing that if I have tuned in settings for the requisite amount of support and know the model doesn't need reorienting and such. This would be useful in OctoPrint for instance.

bubnikv, 2019-01-07:
We took the C++ implementation from upstream, where the conversion to C++ was not finalized yet. Unfortunately the code is not working as expected.

It will likely work if you give it a --nogui parameter.

bubnikv, 2019-01-08:
By the way, apart from the geometry files (.stls, .amfs, .3mfs) what parameters do you use? Which parameters do you use commonly and which rarely?

jayd54, 2019-01-08:
I am using stls and ini files.inwant to generate one gcode from the three stl using command line, will I able to do it ? Or this will be possible in future. Realease.

bubnikv, 2019-01-08:
What ini files are you using? Those generated from the Slic3r UI, therefore
containing all print/filament/printer settings, or are you loading the ini
files of the print/filament/printer profiles?

jayd54, 2019-01-08:
I am using config.ini file which are exported from the SLIC3R UI.

jayd54, 2019-01-08:
Command line aruguemets as per following.

slic3r-console.exe --load config.ini 3C_Lizard_1.stl 3C_Lizard_2.stl 3C_Lizard_3.stl --MERGE

jayd54, 2019-01-22:
Can you please tell me the approximate date of the stable release.

OtterNas3, 2019-01-24:
I cannot get the command line to output correct gcode

If i use this command:
slic3r-console.exe --load .\profiles\OtterMat_HighQualy.ini "D:\Eigene Dateien\3D Druck\### FIRMWARE ###\### Calibration ###\20mm cube.STL"

i get this: 
error: command not supported

However if i add a --slice liek this:
slic3r-console.exe --slice --load .\profiles\OtterMat_HighQualy.ini "D:\Eigene Dateien\3D Druck\### FIRMWARE ###\### Calibration ###\20mm cube.STL"

I get a confirmation:
90 => Exporting G-code to D:\Eigene Dateien\3D Druck\### FIRMWARE ###\### Calibration ###\20mm cube.STL.gcode

However the gcode does not contain any printing information of the model itself only the general informations, the custom start/end g-codes and profile informations are included
See:
`; generated by Slic3r Prusa Edition 1.42.0-alpha3+win64 on 2019-01-24 at 23:32:05

; 

; external perimeters extrusion width = 0.24mm
; perimeters extrusion width = 0.24mm
; infill extrusion width = 0.48mm
; solid infill extrusion width = 0.40mm
; top infill extrusion width = 0.24mm
; support material extrusion width = 0.40mm
; first layer extrusion width = 0.48mm

M73 P0 R0
M201 X1500 Y1500 Z100 E10000 ; sets maximum accelerations, mm/sec^2
M203 X400 Y400 Z12 E80 ; sets maximum feedrates, mm/sec
M204 P1000 R1000 T1000 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X10.00 Y10.00 Z0.40 E5.00 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
;-------
; Start |
;-------

G90; 				Set Positioning to absolute values
M82; 				Set Extruder to absolute values
G21; 				Set Units to Millimeter
M106 S0; 				Turn Off Fan
M900 K0.13; 			Set Linear-Advance factor
M117 Pre-Heat Extruder; 		Send Text
M104 S145; 			Pre-Heat Extruder to 145 to peel off Filament
M76; 				Pause Print Timer
G28 X0 Y0; 			Home X & Y Axis
G0 X0 Y0 Z5 F4000.0; 			Move to Origin
M117 Heating Printbed; 		Send Text
M140 S60; 			Set Bed Temp
M190 S60; 			Wait for Bed to reach Temp
M117 Heating Extruder; 		Send Text
M104 S205; 			Set Extruder Temp
M109 S205; 			Wait for Extruder to reach Temp
M117 Getting Homes; 		Send Text
G28; 				Home all Axis
G29; 				ABL

;-----------
; Priming  |
;-----------
G0 Z5 F3000; 			Raise the Nozzle for moving to Start
M400; 				Wait for moves Finish
M117 Priming Nozzle; 		Send Text
G92 E0 ; 				Reset Extruder distance
G0 X0 Y1 Z0.5 F4000.0; 		Move to Start position for first line
M400; 				Wait for moves Finish
M117 Drawing 1st line; 		Send Text
G1 X200 Y1 Z0.5 E20 F1500; 		Draw the first line
[... 234 lines cut ...]
; infill_speed = 30
; overhangs = 1
; perimeter_extruder = 1
; perimeter_extrusion_width = 0
; perimeter_speed = 30
; perimeters = 5
; small_perimeter_speed = 20
; solid_infill_below_area = 70
; solid_infill_every_layers = 0
; solid_infill_extruder = 1
; solid_infill_extrusion_width = 0.4
; solid_infill_speed = 30
; thin_walls = 0
; top_infill_extrusion_width = 0
; top_solid_infill_speed = 20
; top_solid_layers = 5
; wipe_into_infill = 0
; print_settings_id = OtterMat HighQualy
; filament_settings_id = OtterMat
; printer_settings_id = OtterMat
; printer_model = 
; printer_variant = 
; default_print_profile = 
; default_filament_profile = ""
`

ctilley83, 2019-01-30:
Yep. I'd love to have the command line working as well. I'm writing a web based tool and would love to encode a 3mf file and pass that to the slic3r command line interface. Hopefully you guy's won't abandon my use case.

theopensourcerer, 2019-02-22:
As with @ctilley83 I'm also developing a tool that I was hoping to use Slic3r PE for from the cli. I'm not totally reliant on having an actual gcode file at the end of the process but what I really want is to get some easily machine-readable data back after asking Slic3r to slice my file(s). I'm really interested in knowing things like the amount of filament to be used and the estimated printing time etc. (Think of an on-line pricing calculator for 3d printing services).

As 1.41 operates now (from the cli), this info is provided in the resulting gcode file and can be extracted, albeit using some pretty horrible text parsing.

I hope the use-case info is helpful

PS: Seeing as the bug I reported recently, #1827, is specifically mentioned above I guess mine could probably be closed or marked as a dupe of this one.

bubnikv, 2019-03-14:
The 1.42.0-beta command line processing is based on the upstream code by @alexrj and @lordofhyphens, thanks to them.
The command line format is described by https://manual.slic3r.org/advanced/command-line
with some exceptions:
- instead of --help-options we have --help-fff and --help-sla
- we have an additional parameter --loglevel
- we support --export-sla, not --export-sla-svg or --export-svg
- we don't support --cut-grid --cut-x --cut-y --autosave
- configuration bundled inside an 3mf or amf file is used

closed 2019-03-14 by bubnikv