prusa3d/PrusaSlicer #1583: random home travel during prints

Avatarsia, 2019-01-04:
### Version
_Version of Slic3r 1.42.0 alpha 1

### Operating system type + version
windows x64

### Behavior
slic3r makes some wired travel moves to home position

[image]
[A10MA20M Front (Beta3).zip](https://github.com/prusa3d/Slic3r/files/2727544/A10MA20M.Front.Beta3.zip)

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2019-01-04:
This is a duplicate of https://github.com/prusa3d/Slic3r/issues/1541
 and https://github.com/prusa3d/Slic3r/issues/1505
Closing this one.

closed 2019-01-04 by bubnikv

bubnikv, 2019-01-07:
Fixed with 026bc801e33778ebc5500ad0a3aac9437bca181f