prusa3d/PrusaSlicer #1584: When saving a new profile inheriting a MK3 start GCODE cannot be changed

Patrik-svobodik, 2019-01-04:
### Version
1.42alpha2

### Operating system type + version
macOS Mojave 10.14.2

### Behavior
* When creating a profile that inherits settings from MK3 I am unable to change the custom codes, when I change them they do not save the change

Tucker2015, 2019-01-06:
I'm also getting this issue on Mac Mojave 10.14.2

bubnikv, 2019-01-07:
I believe the issues #1584, #1598, #1601, #1604, #1612 are caused by the same change 8d1b854acbbb1686d92cd38a696dab2a97dec00d
"Replaced all wxString.ToStdString() with wxString.ToUTF8().data()
to be sure that the strings are correctly converted to UTF8."

Fixed with 3bd2db8ba6bc8dcdd96cba6cf51c5882fdca6121, closing

closed 2019-01-07 by bubnikv