prusa3d/PrusaSlicer #1586: 1.42.0 Alpha versions - Win10 app not ending when window is closed- creates multiple instances

tjcolejr, 2019-01-04:
### Version
1.42.0-alpha1&2

### Operating system type + version
Win10 64

### Behavior
When I close the application window with the upper right red X or select file>quit, the window closes, but the task remains active and running in the background.
If I open Slic3r again, 2 instances are running and the second remains active when closed as well. Same with 3rd,4th,5th etc.  
I must open task manager and end each task individually.  I just closed 6 instances, each occupying approximately 20MB of memory.

Rowbotronics, 2019-01-04:
I'm using Windows 10 Home 64-bit and I don't see this issue with either of the 1.42.0 alpha versions.

bubnikv, 2019-01-05:
Don't you have a transfer to OctoPrint running?
Do you have the background processing enabled?

bubnikv, 2019-01-05:
I cannot reproduce it on Windows 10 either, even with background processing
enabled.

tjcolejr, 2019-01-05:
I do have Octoprint set up to print from Slic3r, but this occurs no matter if I'm printing or not.  I will try loading Slic3r onto my laptop and see if I can reproduce.  I did not uninstall my previous Slic3r version before installing the alpha versions.  Would that matter?

bubnikv, 2019-01-05:
no. you should be able to run the stable and alpha releases side by side.

tjcolejr, 2019-01-09:
I cannot duplicate this on my laptop, which has the same version of Windows as my Desktop.   Slic3r has never previously been loaded on my laptop.  If I discover what is going on with my desktop, I'll report back.

bubnikv, 2019-01-29:
Can we close this issue?

tjcolejr, 2019-01-29:
I have wiped out my hard drive and loaded Win 10 64 on it and the first program I loaded was Alpha 3.  I Run the program and it shows in task manager as an active process.  When I close the program, it switches to a background process and will remain there until I kill it.  If I run the program and close it a second time, without killing the original background process, I will then have 2 background processes.  
No change with Alpha 4.  Again, this does not happen on my laptop, only on my desktop.

bubnikv, 2019-01-30:
Would you please attach content of the menu Help -> System Info dialog?
This will give us some data on your system.

I suspect you may have an OpenGL driver installed, which does not support
OpenGL 2.0, so Slic3r is loading a software render from the mesa directory,
and there may be issues, where the system driver fights with the mesa
driver.
You may try to copy the content of the mesa directory one level up (to the
main Slic3rPE directory, next to slic3r.exe) if it helps. Please capture
the System Info before copying the mesa driver though.

tjcolejr, 2019-01-30:
Slic3r Prusa Edition
Version:   1.42.0-alpha4+win64
Build:     Slic3rPE-1.42.0-alpha4+win64-full-201901281428

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 17134), 64-bit edition

OpenGL installation
GL version:   4.5.13399 Compatibility Profile Context 15.201.1151.1008
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon HD 6450
GLSL version: 4.40
Installed extensions:
GL_AMDX_debug_output
GL_AMDX_vertex_shader_tessellator
GL_AMD_conservative_depth
GL_AMD_debug_output
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_multi_draw_indirect
GL_AMD_name_gen_delete
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_sample_positions
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trace
GL_AMD_texture_cube_map_array
GL_AMD_texture_texture4
GL_AMD_transform_feedback3_lines_triangles
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_tessellator
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
[... 185 lines cut ...]
GL_EXT_vertex_attrib_64bit
GL_IBM_texture_mirrored_repeat
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KTX_buffer_region
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_explicit_multisample
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_OES_EGL_image
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control

tjcolejr, 2019-01-31:
> This really looks like some goodbye issue of an opengl driver to me. We don't really have a better way to debug it at your computer than to add some good bye tracing.
>
> By the way, you will get a log if you open command line, type
> set SLIC3R_LOGLEVEL=6
> and start slic3r-console.exe
> from the same command line window. Unfortunately I don't expect you to see anything interesting as of now, as we don't have this "good bye" tracing yet.
>
> Interestingly, you are the only person having this issue. You may try to update your opengl driver.

I checked my open gl and it was at 4.5.  I then  updated to the latest from ATI and the issue persists.  I tried moving the mesa file and no joy.  I will be glad to load goodbye tracing and capture a log.

tjcolejr, 2019-01-31:
It is my video card/drivers.  I run two monitors off this one card. One digital and one HDMI.  If I unplug the HDMI, the persistent Slic3r instance issue goes away, but I only have one monitor.  I am attempting to revert to my on board video and run one monitor from the vga port and the other from the digital.  Right now the pc is not happy and won't boot....repairing.
________________________________
From: Tommy Cole <tjcolejr@gmail.com>
Sent: Thursday, January 31, 2019 7:33:54 AM
To: prusa3d/Slic3r; prusa3d/Slic3r
Cc: Author
Subject: Re: [prusa3d/Slic3r] 1.42.0 Alpha versions - Win10 app not ending when window is closed- creates multiple instances (#1586)

I checked my open gl and it was at 4.5.  I then  updated to the latest from ATI and the issue persists.  I tried moving the mesa file and no joy.  I will be glad to load goodbye tracing and capture a log.

bubnikv, 2019-02-01:
Thaks a lot for your findings. Now we have one less issue to worry about.

bubnikv, 2019-03-02:
It was a graphics card problem, closing.

closed 2019-03-02 by bubnikv