prusa3d/PrusaSlicer #1587: object move only allows  one decimal precision

Mike4U, 2019-01-04:
Slic3r Prusa Edition
Version: 1.41.2+
Build: Slic3rPE-1.41.2+full-201811221501
Platform:
osname=darwin
osvers=13.4.0

Behavior
When trying to move objects using the sliders or slider box in the Settings dialog only one decimal place is used.
~~Note, this may not be helped by fact that my CAD models are in thousands of an inch and must be 2540 times expanded at addition.~~  More of an issue when inches are used ie .1 inch resolution means 
inch user resolution is only 2.54mm

Steps needed to reproduce the problem
1. add model
2. scale 2540, but it doesn't matter, just end up with inch
 size
3. double click object to get settings dialog.
4. click on first object below 'Object' in list and set type as modifier.  Note this issue is only meaningful for MM models or when using modifiers on regular models.
5. type 0.1 in x move slider box.  OK part moves.
6. without hitting return put another digit like 5 so reads 0.15.   BAD part moves no further.

[problem 2.zip](https://github.com/prusa3d/Slic3r/files/2728099/problem.2.zip)

lukasmatena, 2019-01-04:
Could you please test 1.42-alpha2 release? There are some changes that might help you.

Mike4U, 2019-01-04:
thanks
Wow that's different.  I'll have to find the instructions.  Solves the moving problem in a way I don't understand but eliminates using one part as a modifier of other parts color?

Mike4U, 2019-01-05:
OK,  I figured out you have to be in expert mode to use modifiers or do anything useful with MM.  Still don't understand why position of Z isn't 0.0  

But you can close this issue, 

thanks much

bubnikv, 2019-01-05:
> Still don't understand why position of Z isn't 0.0

Z refers to the center of your object.

Closing this issue, at it refers to a feature of 1.41, which was redone in 1.42.

closed 2019-01-05 by bubnikv