prusa3d/PrusaSlicer #1588: Printer chosing bug

martinowren, 2019-01-04:
### Version
1.42.0.alpha2

When you chose 0.6MM or 0.25 nozzle printer profiles, you will get transfered to the SLA profile.

bubnikv, 2019-01-05:
I cannot reproduce it.

Are these 0.6mm and 0.25mm profiles yours, or the Prusa3D system profiles?

Would you please attach the content of your Slic3rPE-alpha config directory?

martinowren, 2019-01-05:
Hi! The ones I found the bug with, are the Prusa3D system profiles. Cannot find a config directory, but I will attach the config bundle (The closest I can find)
[Uploading resources.zip…]()

martinowren, 2019-01-05:
[resources.zip](https://github.com/prusa3d/Slic3r/files/2729503/resources.zip)

bubnikv, 2019-01-07:
Sorry, I need your Slic3rPE-alpha config dir, which you will open from Help menu -> Show Configuration Folder

martinowren, 2019-01-08:
[Slic3rPE-alpha.zip](https://github.com/prusa3d/Slic3r/files/2738452/Slic3rPE-alpha.zip)