prusa3d/PrusaSlicer #1590: alpha2 bug - Moving objects in the preview window is not fluid.

EXOgreen, 2019-01-04:
### Version

1.42.0-Alpha2+win64

### Operating system type + version

Windows 10

### Behavior
When moving any size of model in the preview window the model no longer moves smoothly. alpha1 and before dragging a model or its control arms would smoothly move the model over the buildplate but now it jumps around the buildplate.

_Is this a new feature request?_

no, bug

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

any file

cookew, 2019-01-05:
I can confirm this does happen. It seems to happen when Slic3r's has to make updates to the "Object manipulation" X, Y, and Z coordinates on the right side of the window. If the Slic3r window is shrunk down small enough (essentially making it almost useless) and move something around, dragging items around is fluid as one would expect.

I am running an Intel Core i7-6850K (3.6 GHZ), 128 GB of RAM, and an Nvidia 1080 GTX Graphics card, so I don't think it is a problem with insufficient processing power. Watching the Windows task manager while crazily moving an object around with the window showing the "Object manipulation" stuff and not showing it yeilded about the same CPU load, around 10% for this CPU and around .5%  for the GPU.

Attached is the STL I used.
[test.zip](https://github.com/prusa3d/Slic3r/files/2729022/test.zip)

EXOgreen, 2019-01-05:
This is weird because it was updating fine in alpha1 (I double checked) but they are updating wxwidgets

bubnikv, 2019-01-10:
Should be fixed in the master. Closing.

closed 2019-01-10 by bubnikv