prusa3d/PrusaSlicer #1592: 1.42.0-alpha 1 and 2 missing layers

cookew, 2019-01-04:
### Version
Slic3rPE-1.42.0-alpha1+win64-full-201812231738.zip
Slic3rPE-1.42.0-alpha2+win64-full-201901041551.zip
Both versions were the pre-compile versions from this GitHub repo.
https://github.com/prusa3d/Slic3r/releases

### Operating system type + version
Windows 10 64 bit

### Behavior
Be sure to check the "3D editor view" before printing so you don't end up with missing layers. I noticed this issue after I printed the newer R3 z-axis-bottom.stl for the MK3.

I used all stock settings for the print/filament/printer tabs. The number of layers missing seems to randomly change when it re-renders the preview or outputs the gcode, but also looks like it is generally in the same spots. The missing layers are not visible in the normal "3D editor view". It also seems to only be present when the "0.20mm FAST MK3" print setting is selected.

Just to make sure the STL wasn't messed up, I did download the SCAD file used to generate it and went through the code to see if there was anything weird in there and there isn't. I had even re-generated the STL from the SCAD file I downloaded and it does the same thing.

#### STL/Config (.ZIP) where problem occurs
https://github.com/prusa3d/Original-Prusa-i3/blob/MK3/Printed-Parts/scad/z-axis-bottom.scad
https://github.com/prusa3d/Original-Prusa-i3/blob/MK3/Printed-Parts/stl/z-axis-bottom.stl
[image]

bubnikv, 2019-01-05:
There seems to be a regression issue when going from 1.41.1 to 1.41.2 and 1.42.0, there were fixes done on the stable branch, which seem to be lost in the current master, though I tried hard not to.

bubnikv, 2019-01-07:
This is a duplicate of https://github.com/prusa3d/Slic3r/issues/1562