prusa3d/PrusaSlicer #1593: 1.42.0-alpha 1 and 2 Missing Support Settings in "Print Settings" Tab

3Delights, 2019-01-05:
### Version
Slic3rPE-1.42.0-alpha2+win64-full-201901041551

### Operating system type + version
Windows 10 Pro x64 fully updated

### Behavior
When selecting the "Print Settings" Tab and enabling the Generate Support Option I expect to see all the adjustable settings as in previous versions of Slic3rPE, instead the tab remains blank!

#### Expect:
[image]

#### Actual:
[image]

### STL/Config (.ZIP) where problem occurs
Fresh install from zip, default config and any STL...

### Question
Is this correct behaviour because of it being in Alpha?  Will it be corrected in the Beta/Final releases?

cookew, 2019-01-05:
Did you try changing the Configuration -> Mode menu item setting? Simple is default and only shows the settings with the green to the left of them. Yellow is Advanced and Red is Expert.
[image]

lf-, 2019-01-05:
I think this is also a UI issue which the developers really need to fix, this setting needs to be shown in the configuration wizard so users see it. Bonus points if it has a <NEW> label on it.

Rowbotronics, 2019-01-05:
I agree that it would be nice if it was added to a first-run wizard of some sort. It took me a minute to figure out why settings were missing, and I had already read the release notes that mentioned the change.