prusa3d/PrusaSlicer #1594: 1.42.0.alpha2: New UI: no tooltips on inactive toolbar items

lf-, 2019-01-05:
### Version
_Version of Slic3r used goes here_
1.42.0-alpha2

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10

### Behavior
* _Describe the problem_
New UI adds buttons with no tooltips when inactive, and with icons which are hard to figure out the meaning of, for instance the one the cursor is on in this screenshot:

[image]

Please fix the tooltips as they are most important on inactive items. Another strategy to consider for fixing descriptiveness of icons is something like what Fusion 360 does: 
[image]

Fusion has a top bar with icons, which are also duplicated in a menu with the same icon and a name, so you can associate the icon with a name. It also has tooltips on the buttons and even neat diagrams if you leave your mouse there long enough:
[image]

* _Steps needed to reproduce the problem_
1. Ask a colleague outside the slic3r development team to identify what the button with a square and an O (or is it a zero? or a Q? I have no idea) does.
2. Encourage them to mouse over it to get a tooltip

* _Expected Results_
Know what the new button does.

* _Actual Results_
User cannot identify the use of the button.

  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Sort of.

bubnikv, 2019-01-29:
Tooltips are now shown on inactive toolbar items as well.

closed 2019-01-29 by bubnikv

lf-, 2019-01-29:
By the way, a side bug is that the split into parts button appears on the simple preset despite not really being possible to use.