prusa3d/PrusaSlicer #1595: 1.42.0-alpha2: New UI: object settings should look and act like a button if it is a button

lf-, 2019-01-05:
### Version
_Version of Slic3r used goes here_
1.42.0-alpha2

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10

### Behavior
* _Describe the problem_
The icon to change object settings in the sidebar needs to be right clicked despite it not doing anything if left clicked or double clicked. This seems sort of like a hack to have a tooltip telling the user that they need to do something not compliant with usual conventions rather than complying with the conventions.

It currently looks like this:
[image]

It makes more sense to have this be a button rather than an image you're right clicking on. The following is a MSPaint mockup of an improved version which indicates to the user that it is a button which opens a dropdown:
[image]

I understand if it is an issue with the UI framework preventing this from being implemented sanely. However, this is really strange and confusing UI, there's no two ways about it.

bubnikv, 2019-03-02:
This is a great suggestion. We already spent a significant time trying to work around the wxWidgets limitations. Maybe an external contributor will help us to solve this issue?

Laserology, 2024-05-10:
This can be closed, it appears to be fixed.

closed 2026-09-29 by prusaslicer-bot[bot]