prusa3d/PrusaSlicer #1597: Slic3r ignores Support "Contact Z distance", especially on "flat" overhangs

Sebastianv650, 2019-01-05:
### Version
1.42.0 alpha 2

### Operating system type + version
Win 10 64bit

### Behavior
First I want to point out this is not a new bug in the new alpha, it's also present in the stable releases.

Slic3r ignores the contact Z distance for supports, therefore fusing to layers they are not supposed to support. This might also be the problem, why your developers found supports generated by the work around described in #102 the be not removeable as this will therefore only work on horizontal bridges. 

This is present on nearly all overhang angles, but gets worse as the angle is getting flatter. It's very visible below 20° to the print bed, here is a 10° example where the red squares indicate the are of overlaping support and layer above which shouldn't be supported by this support layer:
[image]

The Support Z-distance is set to 0.1mm, see also the attached .3mf file. So I expect to see 0.4mm (nozzle diameter) + 0.1mm (distance) - 0.2mm (layer height) = 0.3mm gap between the layer to support and the supporting layer. Which is what we see in the screenshot in the not-marked areas. In the red-marked areas we have a real 0.1mm gap, which is way too low so the support fuses to the model in a not-removeable way in this areas.

Also the horizontal XY separation is well above the set value which is 0.3mm in this example but I also tried the default 60%, therefore it's not clear to me what's the reason behind this behaviour expect it's a pure bug.

#### STL/Config (.ZIP) where problem occurs
20° Overhang. Scale Z to 50% to show the bug in extreme: [Keil_20Grad.txt](https://github.com/prusa3d/Slic3r/files/2729247/Keil_20Grad.txt) (rename to .stl)
3mf file with 10° angle, same as used for the screenshot above: [Keil_20Grad.txt](https://github.com/prusa3d/Slic3r/files/2729257/Keil_20Grad.txt) (rename to .3mf)

Sebastianv650, 2019-01-05:
For comparison here is the same 10° example generated with Cura, except I had to use 0.2mm real Z distance there as Cura only supports multiple layer heights for support. And there are no nice colors as I had to use Pronterface for the visualisation, but you should get the idea:
[image]

Each support layer is exactly starting below the layer it will support (depending on XY separation distance of course).

bubnikv, 2019-01-05:
So you consider this to be correct? What I see is a ZERO separation along
the green line.

bubnikv, 2019-01-05:
In regard to the initial screenshot, it is a bug if you are right. Though it is difficult to judge just from a side view, as the support extrusions may seem to overlap looking from the side, but they may not overlap when looking from the top. For this and other things, having a tool in Slic3r to clip the view by a X, Y or Z plane would be helpful.

Sebastianv650, 2019-01-05:
> What I see is a ZERO separation along the green line.

That's true, it's not perfect. But Slic3r has an overlap instead of a "0-contact", so that's what I'm worried about more ;)
This is also not something I found in theory looking at those images, but the result from real failed prints with such overhangs where I wasn't able to remove the supports. This is only the final result of my research why it fused even with a 0.2mm gap (as it is entered as Z-gap in Slic3r) compared to Cura and other gcode testing.

> Though it is difficult to judge just from a side view, as the support extrusions may seem to overlap looking from the side, but they may not overlap when looking from the top.

Beeing an engineer I'm very well aware of that issue, that's why you will see a support enforcer inside the .3mf file. This way, Slic3r isn't creating some zig-zag lines beside the model itself which would mess up the side view. What you see here is what you also see from the top view.

> The extrusions you are showing at the preview are just an approximation, dictated mostly by the need to reduce the number of triangles sent to the graphics card.

True, and I'm not doing pixel measurements with it. Just comparing big gaps to small gaps, namely the vertical gap in the first screenshot showing a "small" gap in the red marked area (exactly 0.1mm according to the gcode) and a "big" one beside the marked area (0.3mm according to gcode).

Sebastianv650, 2019-05-24:
As now PrusaSlicer 2.0 is released, I want to give this a head-up combined with a simple 3mf file for easy confirmability.

- Open the attached 3mf file.
- Note that at layer 39 we have the true top support surface at 4.70mm, resulting from "to-support part layer height" - "nozzle diameter" - "contact Z distance" = 5.2 - 0.4 - 0.1.
- At layer 41, Slicer creates another support layer at the edges only, this one at height 4.90mm. Therefore it fuses with the part, as this correspondes to a Z gap distance of  **-0.1mm** (negative)!
- Switch between Layer 42 and 43 (here the bridge itself is generated) and press "1" to look top down. Note that the perimeter is just above the too high support layer 41, thats where it fuses.

[Bridge_Z-Dist.zip](https://github.com/prusa3d/PrusaSlicer/files/3215631/Bridge_Z-Dist.zip)

As this affects quite a core function from Slicer in my oppinion, especialy as we have now also the function to fine control supports with easy to control modifier meshes, I hope it might raise in the todo list a little bit mabe.

bubnikv, 2019-05-24:
If you read the public comments, the Undo / Redo is the VERY MOST IMPORTANT
FEATURE EVER :-)
So the supports will come after the Undo / Redo.

mat100, 2020-01-04:
some progress in this problem ?

bubnikv, 2021-09-23:
We have implemented a way to disable "thick bridges" in PrusaSlicer 2.4.0-alpha1. Please read the change log.
The supports are now configurable to use similar logic and to produce similar results as the "other" slicers.

closed 2021-09-23 by bubnikv