prusa3d/PrusaSlicer #1599: 1.42.0-alpha2 - linux compile issue - error: no matching function for call to ‘wxString::FromUTF8(const string&)

mkyral, 2019-01-05:
### Version
1.42.0-alpha2

Trying to compile the new version on Gentoo with x11-libs/wxGTK-3.0.4 and it fails on:

```
[ 71%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/GLTexture.cpp.o
cc1plus: warning: /var/tmp/portage/media-gfx/slic3r-pe-1.41.92/work/Slic3r-version_1.42.0-alpha2/build/src/slic3r/libslic3r_gui_pch/pchheader.hpp.gch/.c++: created and used with different settings of -fpie
/var/tmp/portage/media-gfx/slic3r-pe-1.41.92/work/Slic3r-version_1.42.0-alpha2/src/slic3r/GUI/GLTexture.cpp: In member function ‘bool Slic3r::GUI::GLTexture::load_from_file(const string&, bool)’:
/var/tmp/portage/media-gfx/slic3r-pe-1.41.92/work/Slic3r-version_1.42.0-alpha2/src/slic3r/GUI/GLTexture.cpp:40:49: error: no matching function for call to ‘wxString::FromUTF8(const string&)’
  if (!image.LoadFile(wxString::FromUTF8(filename), wxBITMAP_TYPE_PNG))
                                                 ^
In file included from /usr/include/wx-3.0/wx/memory.h:15:0,
                 from /usr/include/wx-3.0/wx/object.h:19,
                 from /usr/include/wx-3.0/wx/event.h:16,
                 from /usr/include/wx-3.0/wx/app.h:19,
                 from /var/tmp/portage/media-gfx/slic3r-pe-1.41.92/work/Slic3r-version_1.42.0-alpha2/build/src/slic3r/libslic3r_gui_pch/pchheader.hpp:95,
                 from <command-line>:0:
/usr/include/wx-3.0/wx/string.h:1742:21: note: candidate: static wxString wxString::FromUTF8(const char*, size_t)
     static wxString FromUTF8(const char *utf8, size_t len = npos)
                     ^~~~~~~~
/usr/include/wx-3.0/wx/string.h:1742:21: note:   no known conversion for argument 1 from ‘const string {aka const std::__cxx11::basic_string<char>}’ to ‘const char*’
make[2]: *** [src/slic3r/CMakeFiles/libslic3r_gui.dir/build.make:412: src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/GLTexture.cpp.o] Chyba 1
make[1]: *** [CMakeFiles/Makefile2:1050: src/slic3r/CMakeFiles/libslic3r_gui.dir/all] Chyba 2
make: *** [Makefile:141: all] Chyba 2

```

I was able to pass this issue by adding `.c_str()` to the `filename` variable.

Then it failed on

```
/var/tmp/portage/media-gfx/slic3r-pe-1.41.92/work/Slic3r-version_1.42.0-alpha2/src/slic3r/GUI/GUI.cpp: In function ‘wxString Slic3r::GUI::from_path(const boost::filesystem::path&)’:
/var/tmp/portage/media-gfx/slic3r-pe-1.41.92/work/Slic3r-version_1.42.0-alpha2/src/slic3r/GUI/GUI.cpp:323:54: error: no matching function for call to ‘wxString::FromUTF8(std::__cxx11::string)’
  return wxString::FromUTF8(path.string<std::string>());
                                                      ^
In file included from /usr/include/wx-3.0/wx/memory.h:15:0,
                 from /usr/include/wx-3.0/wx/object.h:19,
                 from /usr/include/wx-3.0/wx/event.h:16,
                 from /usr/include/wx-3.0/wx/app.h:19,
                 from /var/tmp/portage/media-gfx/slic3r-pe-1.41.92/work/Slic3r-version_1.42.0-alpha2/build/src/slic3r/libslic3r_gui_pch/pchheader.hpp:95,
                 from <command-line>:0:
/usr/include/wx-3.0/wx/string.h:1742:21: note: candidate: static wxString wxString::FromUTF8(const char*, size_t)
     static wxString FromUTF8(const char *utf8, size_t len = npos)
                     ^~~~~~~~
/usr/include/wx-3.0/wx/string.h:1742:21: note:   no known conversion for argument 1 from ‘std::__cxx11::string {aka std::__cxx11::basic_string<char>}’ to ‘const char*’
make[2]: *** [src/slic3r/CMakeFiles/libslic3r_gui.dir/build.make:568: src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/GUI.cpp.o] Chyba 1
make[1]: *** [CMakeFiles/Makefile2:1050: src/slic3r/CMakeFiles/libslic3r_gui.dir/all] Chyba 2
make: *** [Makefile:141: all] Chyba 2

```

Again fixed by adding `.c_str()` to `path.string<std::string>()`

And then again:

```
/var/tmp/portage/media-gfx/slic3r-pe-1.41.92/work/Slic3r-version_1.42.0-alpha2/src/slic3r/GUI/GUI_App.cpp:518:49: error: no matching function for call to ‘wxString::FromUTF8(const string&)’
  wxDir dir(wxString::FromUTF8(localization_dir()));
                                                 ^
In file included from /usr/include/wx-3.0/wx/memory.h:15:0,
                 from /usr/include/wx-3.0/wx/object.h:19,
                 from /usr/include/wx-3.0/wx/event.h:16,
                 from /usr/include/wx-3.0/wx/app.h:19,
                 from /var/tmp/portage/media-gfx/slic3r-pe-1.41.92/work/Slic3r-version_1.42.0-alpha2/build/src/slic3r/libslic3r_gui_pch/pchheader.hpp:95,
                 from <command-line>:0:
/usr/include/wx-3.0/wx/string.h:1742:21: note: candidate: static wxString wxString::FromUTF8(const char*, size_t)
     static wxString FromUTF8(const char *utf8, size_t len = npos)
                     ^~~~~~~~
[... 47 lines cut ...]
                 from /usr/include/wx-3.0/wx/app.h:19,
                 from /var/tmp/portage/media-gfx/slic3r-pe-1.41.92/work/Slic3r-version_1.42.0-alpha2/build/src/slic3r/libslic3r_gui_pch/pchheader.hpp:95,
                 from <command-line>:0:
/usr/include/wx-3.0/wx/string.h:1742:21: note: candidate: static wxString wxString::FromUTF8(const char*, size_t)
     static wxString FromUTF8(const char *utf8, size_t len = npos)
                     ^~~~~~~~
/usr/include/wx-3.0/wx/string.h:1742:21: note:   no known conversion for argument 1 from ‘std::__cxx11::string {aka std::__cxx11::basic_string<char>}’ to ‘const char*’
/var/tmp/portage/media-gfx/slic3r-pe-1.41.92/work/Slic3r-version_1.42.0-alpha2/src/slic3r/GUI/wxExtensions.cpp: In function ‘wxMenuItem* append_submenu(wxMenu*, wxMenu*, int, const wxString&, const wxString&, const string&)’:
/var/tmp/portage/media-gfx/slic3r-pe-1.41.92/work/Slic3r-version_1.42.0-alpha2/src/slic3r/GUI/wxExtensions.cpp:51:70: error: no matching function for call to ‘wxString::FromUTF8(std::__cxx11::string)’
         item->SetBitmap(wxBitmap(wxString::FromUTF8(Slic3r::var(icon)), wxBITMAP_TYPE_PNG));
                                                                      ^
In file included from /usr/include/wx-3.0/wx/memory.h:15:0,
                 from /usr/include/wx-3.0/wx/object.h:19,
                 from /usr/include/wx-3.0/wx/event.h:16,
                 from /usr/include/wx-3.0/wx/app.h:19,
                 from /var/tmp/portage/media-gfx/slic3r-pe-1.41.92/work/Slic3r-version_1.42.0-alpha2/build/src/slic3r/libslic3r_gui_pch/pchheader.hpp:95,
                 from <command-line>:0:
/usr/include/wx-3.0/wx/string.h:1742:21: note: candidate: static wxString wxString::FromUTF8(const char*, size_t)
     static wxString FromUTF8(const char *utf8, size_t len = npos)
                     ^~~~~~~~
/usr/include/wx-3.0/wx/string.h:1742:21: note:   no known conversion for argument 1 from ‘std::__cxx11::string {aka std::__cxx11::basic_string<char>}’ to ‘const char*’

```

Then Slic3r finally builded.

lukasmatena, 2019-01-05:
We currently use wxWidgets 3.1, which already contain the wxString::FromUTF8(const std::string&) overload (unlike 3.0.4 that you link against). I would suggest updating to 3.1. Fixing this one thing may just make you encounter something else that will be more difficult to debug. That is at least my opinion.

bubnikv, 2019-01-05:
> Ok, but in this case you should check for minimal version of wxWidgets in
> configuration part.
>
> I have the same issue (wrong version) with Eigen library as well.

that is a valid point. we shall update the nuild system to check for the
lib versions.

commit cde0aa44 (2019-01-09): Fix: wxWidgets 3.0 back-compat
Fix #1599
closed 2019-01-09 by vojtechkral