prusa3d/PrusaSlicer #1600: Post processing scripts not launched

tomvandeneede, 2019-01-05:
### Version
_Version 1.42.0 alpha 1+_

_What OS are you using OS-X 10.14.2

### Behavior
* _Post processing script under Print settings/Output Options/Post processing script does not seem to execute.  message showing in the status bar: execve failed: format error

* took a working setup from previous version pointing to an executable script in my path, includes no spaces or arguments

* _EI expect the script to be executed after GCode has been executed but script does not get triggered
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

_Is this a new feature request? No

tomvandeneede, 2019-01-05:
Just an sh file that copies the file over to another directory
it used to work with the previous version, it takes the filename as only argument
there are no spaces in the path

tomvandeneede, 2019-01-06:
At my pc now, this is the copy of the script that basically just copies the file to another folder which is a shared flashAir drive in the printer.   Depending on the printer I select I have a different folder.  This way the print loads onto the flashAir disk without having to manually select the disk at each file save... 

script is called p2fa.sh (/Users/......./scripts/p2fa.sh) and is enabled for execution (chmod 755)

#/bin/sh
cp $1 /Volumes/printer_folder

This worked great with the previous version but does not work for me now

lukasmatena, 2019-01-06:
I'm really, really far from being a scripting expert, but the first line in your script appears to be a comment. Unless you change it to `#!/bin/sh`, it does not do anything. And I believe it could lead exactly to the error message that you see (the OS not knowing what it got and what to do with it). But I may be wrong.

tomvandeneede, 2019-01-06:
Sorry my wrong when copying the lines over script has a ! in it.  as said it is the same script that worked before with the previous versions.   I exported the config and loaded it in the new version of slicer to make sure I has all parms for the printers I have

tomvandeneede, 2019-01-06:
Error was due to an issue with the virus checker needing update.  All is working as should

tomvandeneede, 2019-01-06:
Error was due to an issue with the virus checker needing update. All is working as should now

closed 2019-01-06 by tomvandeneede