prusa3d/PrusaSlicer #1602: Feature Request: Estimated print time without exporting G-Code

leopd, 2019-01-05:
### Version
Using Slic3r PE 1.41.2+

### Operating system type + version
On OSX 10.13.6

### Behavior
One of the key trade-offs in 3D printing is print-time vs print-quality.  Slic3r's estimated print time gives us an excellent way to evaluate these trade-offs without using plastic.  But right now AFAICT I can only get the time estimate by exporting a .gcode file.  So I find myself spending a lot of time trying out different settings, slicing, and exporting gcode just to see how much faster/slower those settings will be.  I'm guessing I'm not the only person to do this.

I assume there's some extra processing that has to happen to generate the time estimate, and that this is done during the gcode generation.  If this processing is minimal, it would be nice to do this automatically during slicing, so the time estimate is available immediately after slicing.  If the processing required is substantial, then maybe an explicit button to "estimate print time" without exporting the file.

Rowbotronics, 2019-01-05:
The latest alpha (1.42.0) has this feature already.

lukasmatena, 2019-01-05:
As mentioned above, this is already implemented in the latest alpha release. You don't even have to explicitly slice with (now working) background processing on.

Closing.

closed 2019-01-05 by lukasmatena

leopd, 2019-01-06:
Sweet. Thanks!!