prusa3d/PrusaSlicer #1605: support for uploading to repetier-server

ddavidebor, 2019-01-05:
Hi guys,

It would be very nice if future versions of Slic3r PE would include the possibility of sending files to repetier-server.

The API is quite simple, as can be inferred from the curl command that can be used to accomplish this task:

`"curl" -i -X POST -H "Content-Type: multipart/form-data" -H "x-api-key: <Your API>" -F "a=upload" -F "filename=@[output_filepath]" "http://<Your IP>:3344/printer/model/<Your Printer Name>"
`
The variables required are:

* api key
* host:port
* printer name
* path to the temp file

Regards,
Davide

vojtechkral, 2019-01-11:
I'm hearing that, but please note that Repetier support won't be implemented in the 1.42 release, which is now in alpha and we want to focus on bugfixing only. It could be done in the next release.

capi92, 2019-02-25:
Is there any woarkaround solution waiting for release integration to perform the upload via post-processing scripts in output options of print settings ? We have curl and the syntax, it's just about creating a script, isn't it ?

capi92, 2019-02-26:
This is working, waiting for final implementation:

https://forum.repetier.com/discussion/6257/upload-to-repetier-from-slic3r-prusa-edition?new=1

vojtechkral, 2019-03-04:
Does anyone know if repetier-server has network discovery via Bonjour / Avahi like Octoprint?

ddavidebor, 2019-03-04:
Yes it does have bonjour

vojtechkral, 2019-03-04:
Can you provide a dump from Avahi Zeroconf Browser or equivalent where it's apparent how it advertises itself on the network?

ddavidebor, 2019-03-04:
Hello,
This is what i was able to gather, is it sufficient?

_http._tcp. (World Wide Web HTTP)
repetier-fermiumlabs.local.
192.168.100.82:80
[fe80::250f:10b9:c9b1:e6bf]:80

repetier-fermiumlabs is the custom "server name" that can be set under "settings, connectivity, change server name"

vojtechkral, 2019-03-05:
> This is what i was able to gather, is it sufficient?

Thanks, yes, more or less. Additionally I'd like to know if there's a specific Service Name or something that would distinguish the repetier server from a generic HTTP service.

ddavidebor, 2019-03-05:
I'm not on the development team, but I would guess there is not... it is a generic http service :-) 
Altough, please note, there may be more than one repetier-server on the network.

capi92, 2019-03-05:
> > This is what i was able to gather, is it sufficient?
> 
> Thanks, yes, more or less. Additionally I'd like to know if there's a specific Service Name or something that would distinguish the repetier server from a generic HTTP service.

looking at index.html, I can see that 👍  
<meta name="description" content="Repetier-Server Pro for 3d printer"> 
<meta name="author" content="Roland Littwin">

Which is I guess specific to Repetier...
Please find attached index.html file converted in txt file.

[index.txt](https://github.com/prusa3d/Slic3r/files/2933267/index.txt)

Hancoki, 2019-05-22:
The PrusaSlicer 2.0 is very well done. I'm also using Repetier Server and would like to have a native integration for uploading directly to the server like OctoPrint.

Calling an external script won't work for me.

Best regards

cosmith71, 2019-05-27:
Another vote for integration.  I can't get the script to work and Repetier Server is awesome.

Avatarsia, 2019-08-17:
Push, please add it!

ddavidebor, 2019-10-06:
Hello, As of this month this feature is not really needed as Repetier has introduced a new software called repetier-monitor that allows multiple printer management AND automatic upload from a monitored directly. 

It's indeed working super well, just press "save gcode" and then enter. two clicks.

David

cosmith71, 2019-10-06:
@ddavidebor, I disagree.  I've would much rather have the functionality integrated into PrusaSlicer than having to run another piece of software.  That integration is the only thing I miss about Octoprint.

bubnikv, 2020-02-06:
We welcome pull requests.

cosmith71, 2020-02-07:
Anyone care to take this on?  It's far beyond what I am capable of.  I looked at the recent pull request to add AstroPrint for some clues but PrusaSlicer is far too complex for me and my limited experience to comprehend.

Repetier-Monitor works for my purpose, but I really don't need the extra features (which are still a little buggy) and starting up an extra program is a pain.

kevincraig, 2020-04-30:
I am going to attempt this, I'm a C#/React/Python dev and have done some C++ may take some time since I'll have to brush up on my C++ skills a little more, that said I have Repetier-Server and some time on my hands.

bubnikv, 2020-10-28:
Integrated with https://github.com/prusa3d/PrusaSlicer/pull/4384
It will be part of PrusaSlicer 2.3.0-alpha3.
Closing.

closed 2020-10-28 by bubnikv