prusa3d/PrusaSlicer #1606: HDPI Issues (missing dropdown for MMU extruder selection?)

RJ-Make, 2019-01-06:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

Version 1.42.0 Alpha2 64 Bit
:Note Why not make the version information on the about form copyable?

### Operating system type + version
_What OS are you using, and state any version #s_

Windows 10 36 Bit Version 1803 Build 17134.472

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

The extruder dropdown box is missing (I saw it once), so you cannot select the extruder. 

I'm running 2 4K monitors, and running into some strange intermittent UI glitches. 
When selecting (single click) to load an STL through the menu system, the enitre program's UI shrinks to 1/4 of the screen, and all UI elements are unreadable. 

[image]

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

cosmith71, 2019-01-06:
I see the same issue.  Click once to select the line, click again to show the arrow, and one more time to open the drop-down box.  It's giving me arthritis.  

I've had the UI problem before as well (not just on this version).  You'll need to override the HDPI scaling mode to "System" or "System (Enhanced)", whichever works better for you.  It would be nice to have this fixed going forward.

[image]

bubnikv, 2019-01-07:
Please do not mix different issues in a single thread.

We have tried the "Override high DPI scaling behavior", the result was the 3D scene shrinking to 1/4 of its size on a single Windows 10 computer, so we rather disabled it for now.

The "arthritis" issue is annoying, but we have a hard time to work around the limitations of the wxWidgets GUI library.

cosmith71, 2019-01-07:
@bubnikv 

Thanks for your efforts.  You guys have done amazing things with this.

The High DPI fix I suggested is my own personal work-around.  Interestingly, I didn't need it on these new alpha versions.  They worked correctly for me without it for some reason.

bubnikv, 2019-01-07:
thanks for the compliment.

The alphas are compiled against wxWidgets 3.1.1, which should behave better
in regard to HiDPI.

Mike4U, 2019-01-07:
@RJ-Make 
FYI found if you drag like so it copies.  If you drag top to bottom the window closes.

[image]

RJ-Make, 2019-01-14:
Sorry for the delay in response. The only reason I grouped the two is because I original thought it was caused by an HDPI issue. 

No OS HDPI overrides works for my setup, and the trigger is very intermittent. The problem is only present in the Alpha's (all 3). 

@Mike4U That only works if you are on the 'System Info' window. It does not however work in the About box, which is where the topic creation information instruct us to go.

_Use `About->About Slic3r` for release versions_

FidelCapo, 2019-01-15:
Created JIRA 765 for window closing issue.

FidelCapo, 2019-01-16:
Closing System info and About Slic3r dialogs closing issue will be fixed in next release.

bubnikv, 2019-01-29:
> I'm running 2 4K monitors, and running into some strange intermittent UI glitches.
When selecting (single click) to load an STL through the menu system, the entire program's UI shrinks to 1/4 of the screen, and all UI elements are unreadable.

Do you still experience the issue with the Alpha4? We have updated the wxWidgets to the latest release, it may be fixed.

bubnikv, 2019-03-02:
@cosmith71  Please retest with the alpha7 that is to be released today. My hopes are low though. For you, the best solution is likely to override the HiDPI handling as you have already found. The good news is that you seem to be the only person experiencing such an issue on Windows.

cosmith71, 2019-03-02:
@bubnikv will do.  Looking forward to the release.  I tried to compile but was unable to due to errors.

cosmith71, 2019-03-04:
So far so good.  I'll update with any issues.

nebbywan, 2019-03-04:
> I see the same issue. Click once to select the line, click again to show the arrow, and one more time to open the drop-down box. It's giving me arthritis.

I'm on alpha7 x64 but still seeing this sequence as necessary to change the extruder selection. I'm running on a 4K notebook running Win 10 Home v1809 build 17763.316

bubnikv, 2019-03-04:
@nebbywan Sorry, but for now you will have to fight the artritis before we find some work around for the drop down in a table issue. According to the wxWidgets authors (wxWidgets is a multi-platform library that wraps around the native controls of your platform), the behavior is correct: You select the field first, then you drop down the menu.

I am hoping that the HiDPI issues were solved, and this issue was about the HiDPI issues. Closing.

closed 2019-03-04 by bubnikv

bubnikv, 2019-08-23:
We improved the "arthritis" issue somehow:
First the double clicks are disabled on Windows for the side bar and for the 3D scene,
second we removed one click required to activate the extruder combo box in case the line was already active.