prusa3d/PrusaSlicer #1608: modifier meshes (support enforcers etc) are deleted on "Reload from Disk" [1.42.0-alpha1]

simonkuehling, 2019-01-06:
### Version
1.42.0-alpha1+linux64

### Operating system type + version
Ubuntu 18.04 (64bit)

### Behavior
When using "Reload from Disk" for a model on the plater, all previously applied modifier meshes are deleted.
I often use reloading when I see need to repair the STL file externally (through netfabb or similar). It would be super convenient if the modifier meshed would stay in place in order to save some work.

(p.s.: Again, I REALLY like where you are going with the UI improvements. So productive!)

florek41, 2019-01-07:
Additionaly same happens with other settings applied to model - rotation, scaling and selected extruder or custom settings per object.

In my case i use "Reload from disk" when I noticed something after slicing it - ex. I need to make some walls thicker etc.

bubnikv, 2019-01-07:
We are aware that the "Reload from disk" feature is far from being perfect.

bubnikv, 2019-01-08:
Duplicate of https://github.com/prusa3d/Slic3r/issues/1471
closing

closed 2019-01-08 by bubnikv