prusa3d/PrusaSlicer #1609: File-->"Save" and File-->"Save as" behave the same [1.42.0-alpha2+linux64]

simonkuehling, 2019-01-06:
### Version
1.42.0-alpha2+linux64

### Operating system type + version
Ubuntu 18.04 (64bit)

### Behavior
The project file "save" and "save as" menu items are both opening the file location choose dialog.
I would expect that "save" would silently overwrite the currently loaded project file if it was saved previously, whereas "save as" would always open the dialog.

Is this intended behavior or just a bug maybe?

Thanks for your work!

lf-, 2019-01-07:
It would work the same if you haven't saved the file already. Otherwise, definitely a bug.

ddavidebor, 2019-01-16:
Similar issue on mac osx in the Alpha 3: if i open the file (double click from finder), modify then do "save", it asks for the path of the file.

ochm, 2019-02-13:
In the alphha5 the same behavior.

bubnikv, 2019-06-06:
The PrusaSlicer 2.0.0 behaves better, there is an initial implementation of a "current project" concept, the current project name is shown at the application title etc.

closed 2019-06-06 by bubnikv