prusa3d/PrusaSlicer #1610: LAYER EDITING DOES NOT WORK

DROBORD, 2019-01-06:
### Version
_Version of Slic3r 1.42.0

_### Operating system WIN10 64bit

### Behavior
Layer editing does not work - it shows that the layer curve (blue) changes, but it does not remain changed after releasing mouse button. Valid for LEFT and RIGHT mouse buttons

Any object selected.

ram64, 2019-01-06:
I have the same problem. You can see the layer changes being done in both the model and the curve view but a second after releasing the mouse button everything resets to default. I have tried with background processing on and off and it makes no difference. I am on Windows 10 x64.

bubnikv, 2019-01-21:
Fixed with d64b55d27d1716cb3f5ead25c269dce73028899a and ceb08d6380ea8bc64516dae55e896c8b8b67836e
It will be contained in the following alpha release. 
Closing.

closed 2019-01-21 by bubnikv